Ultimaker/Cura #1131: gcode {keywords} reference?

filipgoc, 2016-11-10:
Hi, 

In the old days, Cura supported things like **{date}, {time} and {filament_cost}...**
NOT ANYMORE. 

But some like **{material_print_temperature} or {layer_height}** still work. 

### Where/ how can I find what {keywords} are available to me when crafting start gcode script? 

Thanks!

fieldOfView, 2016-11-10:
See the "settings" in fdmprinter.def.json
https://github.com/Ultimaker/Cura/blob/master/resources/definitions/fdmprinter.def.json

filipgoc, 2016-11-10:
Thanks @fieldOfView 
All of those do indeed work. 

The only question is then where would I get reference to keywords that are not part of the machine definition? 

filament cost, cost of the sliced object, time it takes to print, when the stuff was sliced... 

Any ideas? Thanks.

Ghostkeeper, 2016-11-11:
They are not implemented at the moment. This functionality was lost when Cura's front-end was rewritten for version 2.

filipgoc, 2016-11-11:
Thanks @Ghostkeeper 
Noted.

Drayson80, 2016-11-11:
Is there a plan to bring them in again? It´s easier to check time, cost and material use on "old" gcode files when reprinting than search for the part to re-slice...

Ghostkeeper, 2016-11-11:
Yeah. I've added an issue to our internal tracker. It's not accepted for any sprint yet, but I'm sure that the people who do the planning will think it's a worthwhile addition. It shouldn't be very hard to add.

If one of you would like to help out with a pull request, it would probably need to be added to this bit of code: https://github.com/Ultimaker/Cura/blob/master/plugins/CuraEngineBackend/StartSliceJob.py#L256

nickthetait, 2016-11-11:
I just assigned my team (Aleph Objects) to work on this. https://code.alephobjects.com/T576
Is this a full list of keywords that Legacy Cura supported? https://github.com/daid/LegacyCura/blob/SteamEngine/Cura/util/profile.py#L178

filipgoc, 2016-11-12:
@nickthetait it looks like that's all that legacy cura supported. 
very swift of you. 

However, it'd be awesome to ADD: 
- {quality}
- {material}
- {material_label}
etc.
basically so that the gcode carries information about what slicing presets were used
(both material and general/quality settings)

- icing on the top would be to {3dmodel_filename} or something like that. 
so that the gcode itself would say "3Dbenchy.stl" somewhere on the inside.

Ghostkeeper, 2016-11-12:
These are the tags that Legacy Cura supported: https://github.com/daid/LegacyCura/blob/SteamEngine/Cura/util/profile.py#L1254
A bit further down the file.

nickthetait, 2016-11-15:
Thanks to the both of ya :smiley:

filipgoc, 2016-11-16:
I have an idea. 

What about adding {filename} @nickthetait ?

That way, any user could put **M117 {filename}** in their gcode and know what they are printing under any basic Marlin config! That'd be nice!

{filename} means the gcode filename...

kakaroto, 2016-11-21:
@filipgoc I'm not sure how {filename} would work, because Cura has no control over the actual filename once it's written, it can be modified later by the user, and it could cause confusion if it prints the wrong filename to the user. The output filename can always be used of course.
The {3dmodel_filename} you suggested has another similar problem in that it would have to be a list potentially, in the case of the user loading multiple stl files onto the platform.

filipgoc, 2016-11-21:
@kakaroto good point. 
It makes much more sense to get this into Marlin (or whatever firmware whoever is using... )

I just got lazy and thought this would be nice to have. 

I retract the {filename} suggestion.

Ghostkeeper, 2016-11-22:
Perhaps {jobname} then? To output the job name being displayed in the lower right corner?

Ghostkeeper, 2017-02-18:
The new product manager for Cura at Ultimaker rejected this issue for being too unimportant. This means that the main developers can't spend time on it.

closed 2018-02-27 by ianpaschal

JohnWasser, 2018-03-11:
So the available keywords are: 
    {time}, {date}, {day}
    {print_time}            (legacy, always shows as  '#P_TIME#')
    {filament_amount} (legacy, always shows as '#F_AMNT#')
    {filament_weight}  (legacy, always shows as '#F_WGHT#')
    {filament_cost}       (legacy, always shows as '#F_COST#')
    {profile_string}      (Name of the current profile?)
    F{max_z_speed}       Replaced with 'F' and travel_speed * 60
    F{print_speed}          Replaced with 'F' and named profile setting * 60
    F{retraction_speed}  Replaced with 'F' and named profile setting * 60
    F{travel_speed}         Replaced with 'F' and named profile setting * 60
    F{bottom_layer_speed}  Replaced with 'F' and named profile setting * 60
    F{cool_min_feedrate}    Replaced with 'F' and named profile setting * 60
    {name of any profile setting}       Replaced with profile setting value
    {name of any prefernce setting}  Replaced with preference setting value

Ghostkeeper, 2018-04-05:
No, that's wrong. Available keywords are only:
`{time}`, `{date}`, `{day}`: Time of saving g-code, full date, day of the week.
`{initial_extruder_nr}`: The extruder that the print starts on.
`{print_bed_temperature}`: Bed temperature.
`{print_temperature}`: Nozzle 0 temperature.
`{`name of any profile setting`}`: Replaced with profile setting value.

The rest of the keywords you name don't exist.