Ultimaker/Cura #1134: quality settings per machine - names parsing? (generic_pla_jellybox_0.4_mm)

filipgoc, 2016-11-11:
I'm trying to add custom quality and material profiles for a custom machine (called jellybox)
I'm using ultimaker2_plus as my template. 
Where I'm stuck now are these lines: 
```
[metadata]
material = generic_pla_ultimaker2_plus_0.4_mm
```
from the quality > ultimaker2_plus folder of profiles.

**How and where is it parsed?**
I don't see this specific key (generic_pla_ultimaker2_plus_0.4_mm) repeat anywhere. 
Yet it seems this is the key to the kingdom. 

**So far I have already done:** 
- Custom printer definition including `has_machine_materials` and `has_machine_qualities`
- Custom materials definitions that already show up correcly with my printer
- Added my printer to `__product_id_map` in `XmlMaterialProfile.py` 
- New folder under `resources > quality` with custom quality profile - but don't know what the 'metadata > material' key should be. 

Thanks to anyone for hints in the right direction!
(It's also possible the question is wrong.)

Ghostkeeper, 2016-11-14:
It is a reference to the place _inside_ the material XML file.

The material XML files define different settings for different machines and different nozzle sizes. So essentially, it defines multiple profiles in one file. Cura reads it that way, too. If I'm not mistaken, it is a combination of the material ID (`generic_pla`) and the variant ID (`ultimaker2_plus_0.4_mm`).

It is a very opaque piece of magic there. We are altogether not happy with how the XML profiles turned out, but it was thought at the time to be a good way to share material definitions between Cura and the Ultimaker 3 as well as to set a standard in the industry. For those purposes it works fine, but it's not the easiest file type to implement.

filipgoc, 2016-11-17:
It is pretty magical! (Not in 100% good way)

Is there a current effort to make it less opaque? Or should I give this serious effort because that's the way it's gonna be for a while? 
On sunday or so I may give it another spin...

filipgoc, 2016-11-17:
As always, thanks for an informative answer!

filipgoc, 2016-11-17:
ALTERNATIVELY, 
Is there other acceptable way for me to publish machine specific profiles?

I feel that putting the profile in the general area without having it only for a specific machine is hogging the interface for unsuspected users... 

(That's the whole reason why I'm trying this whole has_machine_materials and has_machine_qualities business. 

Unfortunately it seems noone else other than you ultimaker guys has done it so far...)

awhiemstra, 2016-11-17:
We will not be changing it for 2.4. So your best bet is to stick with it for now.

filipgoc, 2016-11-17:
Thanks @awhiemstra. I'll give it a shot then.

closed 2017-11-14 by ChrisTerBeke