Ultimaker/Cura #1137: "Pause at height" Does not display tool head move

theflyingbean, 2016-11-12:
Cura 2.3.1
Windows 10
Custom printer (Lulzbot Mini)

When Using the "Pause at height" plugin, the tool head moving away from the part and parking is not displayed when in layer view mode. This is a feature that is in Cura 15.xx and is helpful to visualize where your print pause will occur. I am wondering if it could be added to Cura 2. 

Cheers

thopiekar, 2016-11-13:
@nickthetait, @kakaroto Ping

As far as I know this feature is included in the PostProcessing plugin.

BerndJM, 2016-11-13:
Can confirm this for Linux also, so it seems to be OS independent ...
I think to remember that this worked as expected in versions before 2.3.1 - but cannot say exactly when it is gone :-(

thopiekar, 2016-11-13:
@BerndJM So missing in PostProcessingPlugin too?

BerndJM, 2016-11-13:
it seems so

fieldOfView, 2016-11-13:
Anything added or changed by the postprocessing plugin has never shown in the layerview since the 2.x rewrite.

The postprocessing plugin is invoked when saving or sending gcode, not before. So what you see in layerview has not yet been postprocessed

Ghostkeeper, 2016-11-14:
I'm marking this as a bug (it technically is), but for the reason that fieldOfView says, it is probably not getting fixed any time soon.

For version 2.4 we are planning to upgrade the layer view a bit to allow for loading g-code from a file. Perhaps we could consider performing the post-processing a bit earlier then and see what we can do to display the result in the layer view (or whatever view is made for displaying g-code).

fieldOfView, 2016-11-14:
> For version 2.4 we are planning to upgrade the layer view a bit to allow for loading g-code from a file

Have you seen the work by lulzbot here: https://code.alephobjects.com/T466 ?

Ghostkeeper, 2016-11-14:
I hadn't! I'll be sure to link that to our Jira issue too... Nice work over there, guys.

nickthetait, 2016-11-14:
Yeah, bulk of the work for GCodeReader is done. :smile:  It will be going through a code review, then I'll make a PR.
I've not yet tried this plugin in Cura 2.x, so don't have anything else to add.

kakaroto, 2016-11-14:
That makes sense, if the post processing works on the gcode and cura2 doesn't parse the resulting gcode, it can't show it in the layer view. So I assume that once we finish the work on the gcode loader (which would also fix #577), then the layer view would parse the result gcode and show that instead of whatever it is showing currently. That would fix the current issue.

jackha, 2018-07-26:
Marked as deferred because of inactivity

Ghostkeeper, 2018-09-06:
A problem with this is that parsing the g-code takes an order of magnitude longer than displaying the pre-generated polygons that CuraEngine sends to the front-end. We don't want to have the user wait for an additional minute after each slice to display layer view.

closed 2018-11-26 by Appesteijn