Ultimaker/Cura #1138: Print Dialog 15.06.4

BNNorman, 2016-11-12:
Running on Debian (Jessie). Now realised that if the printer is not up to temperature BEFORE the print to USB button is used then the print dialog may or may not display the full buttons. I find if I close the dialog and re-open when the printer is at the required temperature then the Print/cancel buttons are fully visible.

thopiekar, 2016-11-13:
As you are using an older version of Cura, please ask for support at: https://github.com/daid/LegacyCura/issues

Thanks

closed 2016-11-13 by thopiekar

BNNorman, 2016-11-14:
Thank you. I am new to this. I had presumed 15.x was newer than 2.x (dates on the website seemed similar – very confusing to me)– I will download 2.31 and install it. 

 

Closed #1138 <https://github.com/Ultimaker/Cura/issues/1138> .