Ultimaker/Cura #1139: Cura 2.3.1-0 Crashes On Load

sorvis, 2016-11-13:
Today I updated Ubuntu to 16.10 and also probably installed the latest updates of the Cura build from PPA thopiekar as well. Now when I tried to load Cura I see this error message. "_QObject::connect: Parentheses expected, signal MainWindow::_" (Fully loading messages below)

Does this seem like an issue with my system or is this an issue with the latest version?

> 2016-11-12 22:25:32,113 - INFO - .../UM/PluginRegistry.py (loadPlugin [63]): Loaded plugin ConsoleLogger
2016-11-12 22:25:32,123 - INFO - /usr/lib/cura/plugins/CuraEngineBackend/CuraEngineBackend.py (__init__ [40]): Found CuraEngine at: /usr/bin/CuraEngine
2016-11-12 22:25:32,123 - INFO - .../UM/PluginRegistry.py (loadPlugin [63]): Loaded plugin CuraEngineBackend
2016-11-12 22:25:32,132 - INFO - .../UM/PluginRegistry.py (loadPlugin [63]): Loaded plugin RotateTool
2016-11-12 22:25:32,136 - INFO - .../UM/PluginRegistry.py (loadPlugin [63]): Loaded plugin TranslateTool
2016-11-12 22:25:32,140 - INFO - .../UM/PluginRegistry.py (loadPlugin [63]): Loaded plugin MirrorTool
2016-11-12 22:25:32,148 - INFO - .../UM/PluginRegistry.py (loadPlugin [63]): Loaded plugin ScaleTool
2016-11-12 22:25:32,150 - INFO - .../UM/PluginRegistry.py (loadPlugin [63]): Loaded plugin SelectionTool
2016-11-12 22:25:32,153 - INFO - .../UM/PluginRegistry.py (loadPlugin [63]): Loaded plugin CameraTool
2016-11-12 22:25:32,155 - INFO - .../UM/PluginRegistry.py (loadPlugin [63]): Loaded plugin STLWriter
2016-11-12 22:25:32,157 - INFO - .../UM/PluginRegistry.py (loadPlugin [63]): Loaded plugin OBJReader
2016-11-12 22:25:32,159 - INFO - .../UM/PluginRegistry.py (loadPlugin [63]): Loaded plugin OBJWriter
2016-11-12 22:25:32,160 - WARNING - /usr/lib/uranium/plugins/FileHandlers/STLReader/STLReader.py (<module> [5]): Could not find numpy-stl, falling back to slower code.
2016-11-12 22:25:32,161 - INFO - .../UM/PluginRegistry.py (loadPlugin [63]): Loaded plugin STLReader
2016-11-12 22:25:32,166 - INFO - .../UM/PluginRegistry.py (loadPlugin [63]): Loaded plugin 3MFWriter
2016-11-12 22:25:32,166 - WARNING - .../UM/PluginRegistry.py (loadPlugin [63]): Plugin ConsoleLogger was already loaded
2016-11-12 22:25:32,168 - INFO - .../UM/PluginRegistry.py (loadPlugin [63]): Loaded plugin SimpleView
2016-11-12 22:25:32,170 - INFO - .../UM/PluginRegistry.py (loadPlugin [63]): Loaded plugin WireframeView
2016-11-12 22:25:32,173 - INFO - .../UM/PluginRegistry.py (loadPlugin [63]): Loaded plugin LocalFileOutputDevice
2016-11-12 22:25:32,175 - INFO - .../UM/PluginRegistry.py (loadPlugin [63]): Loaded plugin FileLogger
2016-11-12 22:25:32,177 - INFO - .../UM/PluginRegistry.py (loadPlugin [63]): Loaded plugin LayerView
2016-11-12 22:25:32,179 - INFO - .../UM/PluginRegistry.py (loadPlugin [63]): Loaded plugin CuraProfileReader
2016-11-12 22:25:32,181 - INFO - .../UM/PluginRegistry.py (loadPlugin [63]): Loaded plugin VersionUpgrade21to22
2016-11-12 22:25:32,182 - INFO - .../UM/PluginRegistry.py (loadPlugin [63]): Loaded plugin X3DReader
2016-11-12 22:25:32,184 - INFO - .../UM/PluginRegistry.py (loadPlugin [63]): Loaded plugin ChangeLogPlugin
2016-11-12 22:25:32,186 - INFO - .../UM/PluginRegistry.py (loadPlugin [63]): Loaded plugin PerObjectSettingsTool
2016-11-12 22:25:32,187 - INFO - .../UM/PluginRegistry.py (loadPlugin [63]): Loaded plugin ImageReader
2016-11-12 22:25:32,191 - INFO - .../UM/PluginRegistry.py (loadPlugin [63]): Loaded plugin UltimakerMachineActions
2016-11-12 22:25:32,192 - INFO - .../UM/PluginRegistry.py (loadPlugin [63]): Loaded plugin GCodeProfileReader
2016-11-12 22:25:32,194 - INFO - .../UM/PluginRegistry.py (loadPlugin [63]): Loaded plugin XRayView
2016-11-12 22:25:32,194 - WARNING - .../UM/PluginRegistry.py (loadPlugin [63]): Plugin CuraEngineBackend was already loaded
2016-11-12 22:25:32,195 - INFO - .../UM/PluginRegistry.py (loadPlugin [63]): Loaded plugin GCodeWriter
2016-11-12 22:25:32,197 - INFO - .../UM/PluginRegistry.py (loadPlugin [63]): Loaded plugin SolidView
2016-11-12 22:25:32,198 - INFO - .../UM/PluginRegistry.py (loadPlugin [63]): Loaded plugin AutoSave
2016-11-12 22:25:32,200 - INFO - .../UM/PluginRegistry.py (loadPlugin [63]): Loaded plugin CuraProfileWriter
2016-11-12 22:25:32,201 - INFO - .../UM/PluginRegistry.py (loadPlugin [63]): Loaded plugin LegacyProfileReader
2016-11-12 22:25:32,203 - INFO - .../UM/PluginRegistry.py (loadPlugin [63]): Loaded plugin MachineSettingsAction
2016-11-12 22:25:32,209 - INFO - .../UM/PluginRegistry.py (loadPlugin [63]): Loaded plugin USBPrinting
2016-11-12 22:25:32,231 - INFO - .../UM/PluginRegistry.py (loadPlugin [63]): Loaded plugin SliceInfoPlugin
2016-11-12 22:25:32,263 - INFO - .../UM/PluginRegistry.py (loadPlugin [63]): Loaded plugin OctoPrintPlugin
2016-11-12 22:25:32,271 - INFO - .../UM/PluginRegistry.py (loadPlugin [63]): Loaded plugin XmlMaterialProfile
2016-11-12 22:25:32,274 - INFO - .../UM/PluginRegistry.py (loadPlugin [63]): Loaded plugin 3MFReader
2016-11-12 22:25:32,276 - INFO - .../UM/PluginRegistry.py (loadPlugin [63]): Loaded plugin RemovableDriveOutputDevice
2016-11-12 22:25:32,277 - INFO - .../UM/Qt/QtApplication.py (__init__ [39]): Command line arguments: {'external-backend': False, 'file': [], 'debug-mode': False, 'disable-textures': False}
2016-11-12 22:25:32,280 - INFO - .../UM/VersionUpgradeManager.py (upgrade [98]): Looking for old configuration files to upgrade.
2016-11-12 22:25:32,385 - DEBUG - /usr/lib/cura/plugins/OctoPrintPlugin/OctoPrintOutputDevicePlugin.py (_onServiceChanged [138]): Bonjour service added: octopi._octoprint._tcp.local
2016-11-12 22:25:34,229 - DEBUG - .../UM/Backend/Backend.py (_logSocketState [165]): Socket state changed to Listening
2016-11-12 22:25:34,233 - INFO - .../UM/Backend/Backend.py (startEngine [55]): Started engine process: /usr/bin/CuraEngine
2016-11-12 22:25:34,234 - DEBUG - .../UM/Backend/Backend.py (_backendLog [88]): [Backend] Calling engine with: ['/usr/bin/CuraEngine', 'connect', '127.0.0.1:49674', '-j', '/usr/share/cura/resources/definitions/fdmprinter.def.json', '']
2016-11-12 22:25:34,235 - DEBUG - .../UM/Backend/Backend.py (_backendLog [88]): [Backend] 
2016-11-12 22:25:34,235 - DEBUG - .../UM/Backend/Backend.py (_backendLog [88]): [Backend] Cura_SteamEngine version DEV
2016-11-12 22:25:34,236 - DEBUG - .../UM/Backend/Backend.py (_backendLog [88]): [Backend] Copyright (C) 2014 David Braam
2016-11-12 22:25:34,236 - DEBUG - .../UM/Backend/Backend.py (_backendLog [88]): [Backend] 
2016-11-12 22:25:34,236 - DEBUG - .../UM/Backend/Backend.py (_backendLog [88]): [Backend] This program is free software: you can redistribute it and/or modify
2016-11-12 22:25:34,236 - DEBUG - .../UM/Backend/Backend.py (_backendLog [88]): [Backend] it under the terms of the GNU Affero General Public License as published by
2016-11-12 22:25:34,237 - DEBUG - .../UM/Backend/Backend.py (_backendLog [88]): [Backend] the Free Software Foundation, either version 3 of the License, or
2016-11-12 22:25:34,237 - DEBUG - .../UM/Backend/Backend.py (_backendLog [88]): [Backend] (at your option) any later version.
2016-11-12 22:25:34,237 - DEBUG - .../UM/Backend/Backend.py (_backendLog [88]): [Backend] 
2016-11-12 22:25:34,240 - DEBUG - .../UM/Backend/Backend.py (_backendLog [88]): [Backend] This program is distributed in the hope that it will be useful,
2016-11-12 22:25:34,240 - DEBUG - .../UM/Backend/Backend.py (_backendLog [88]): [Backend] but WITHOUT ANY WARRANTY; without even the implied warranty of
2016-11-12 22:25:34,240 - DEBUG - .../UM/Backend/Backend.py (_backendLog [88]): [Backend] MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.  See the
2016-11-12 22:25:34,240 - DEBUG - .../UM/Backend/Backend.py (_backendLog [88]): [Backend] GNU Affero General Public License for more details.
2016-11-12 22:25:34,241 - DEBUG - .../UM/Backend/Backend.py (_logSocketState [165]): Socket state changed to Connected
2016-11-12 22:25:34,242 - DEBUG - .../UM/Backend/Backend.py (_backendLog [88]): [Backend] 
2016-11-12 22:25:34,242 - DEBUG - .../UM/Backend/Backend.py (_onSocketStateChanged [155]): Backend connected on port 49674
2016-11-12 22:25:34,242 - DEBUG - .../UM/Backend/Backend.py (_backendLog [88]): [Backend] You should have received a copy of the GNU Affero General Public License
2016-11-12 22:25:34,242 - DEBUG - .../UM/Backend/Backend.py (_backendLog [88]): [Backend] along with this program.  If not, see <http://www.gnu.org/licenses/>.
[... cut ...]
2016-11-12 22:25:38,450 - DEBUG - /usr/lib/uranium/plugins/FileHandlers/STLReader/STLReader.py (read [35]): Loaded a mesh with 8052 vertices
2016-11-12 22:25:38,718 - DEBUG - .../UM/Qt/Bindings/Theme.py (load [118]): Loading theme file: /usr/share/cura/resources/themes/cura/theme.json
2016-11-12 22:25:38,798 - DEBUG - .../UM/Qt/Bindings/Theme.py (load [118]): Loaded theme /usr/share/cura/resources/themes/cura
QObject::connect: Parentheses expected, signal MainWindow::
QObject::connect: Parentheses expected, signal MainWindow::
QObject::connect: Parentheses expected, signal MainWindow::
Segmentation fault (core dumped)

thopiekar, 2016-11-13:
@sorvis I don't get why it happens. Which desktop do you use?
We already had an user here with the same problem. The reason is that Qt5 is "over-patched" on Ubuntu.
I personally use KDE or Mate and it works very well for me here.

Regards

sorvis, 2016-11-13:
@thopiekar Thank you for the reply.

I'm using Unity 7.5. 

Do you have any suggestions of any other steps I could take to debug the issue further to get it to give a more helpful error message?

sorvis, 2016-11-13:
@thopiekar As a test I logged out and when into Gnome and it does load without crashing. What would the difference be in how it runs in Gnome?

thopiekar, 2016-11-13:
No idea. We already debugged this problem here and there is something Unity seems to use or explicitly enables that makes Cura crash.
In my point of view we should blame it on Canonical - it is simply not Cura related..

sorvis, 2016-11-13:
@thopiekar I was peeking around in my system logs and I did find this at the end of my syslog after I try running Cura. The last message there does seem to be unity specific.

> Nov 13 15:15:36  compiz[17484]: QObject::connect: Parentheses expected, signal MainWindow::
Nov 13 15:15:36  kernel: [ 3443.168670] cura[21990]: segfault at 0 ip 00007f2a9f100c7f sp 00007ffda547ce30 error 4 in libQt5XcbQpa.so.5.6.1[7f2a9f05f000+fb000]
Nov 13 15:15:36  compiz[17484]: message repeated 2 times: [ QObject::connect: Parentheses expected, signal MainWindow::]
Nov 13 15:15:36  unity-panel-ser[17501]: menus_destroyed: assertion 'IS_WINDOW_MENU(wm)' failed

thopiekar, 2016-11-13:
@sorvis Maybe bringing back the menu to Cura solves the problem? http://askubuntu.com/questions/541449/how-do-i-always-show-menu-items-in-unity

sorvis, 2016-11-13:
@thopiekar The setting does not seem to impact it. I still got the same error in the syslog. It was a good idea to try though.

thopiekar, 2016-11-13:
Hmm, sorry. What about running `UBUNTU_MENUPROXY= cura` in the terminal?

sorvis, 2016-11-13:
I ran `UBUNTU_MENUPROXY= cura` and it still crashes with the same errors in the log and terminal. 

I was a bit confused why you didn't say `UBUNTU_MENUPROXY=0 cura` so I tried that as well before running the cura command but that didn't work as well.

awhiemstra, 2016-11-14:
We already set UBUNTU_MENUPROXY=0 in Cura because otherwise you get no menu at all. The only solution we found is to use the AppImage instead of the .deb: http://software.ultimaker.com/current/Cura-2.3.1.AppImage . This does not make use of Ubuntu's broken Qt and thus works correctly.

thopiekar, 2016-11-14:
Well, as this problem is related to my personal PPA and the problem does not occur using AppImage...

closed 2016-11-14 by thopiekar