Ultimaker/Cura #1141: feature request: speed overrides in materials

MarcusWolschon, 2016-11-13:
Some materials have maximum recommed printing speeds (bonus points for doing the same with travel speeds because a too large gap between printing and travel speeds will obviously cause quality issues).
This should be a part of the material profile to override the values from the printing profile
just like a maximum machine speed overrides the speeds in a printing profile that the machine is not capable of reaching.

filipgoc, 2016-11-13:
+1. I'm curious about this holistically. 
(Whether everything in material should override anything in quality https://github.com/Ultimaker/Cura/issues/1135)

Ghostkeeper, 2016-11-14:
The XML materials are not as powerful in what they can express as the rest of the profiles in Cura. Currently it defines only 7 settings (seen here: https://github.com/Ultimaker/Cura/blob/master/plugins/XmlMaterialProfile/XmlMaterialProfile.py#L532).This is because it was designed as an industry-wide standard.

We have plans to extend the profiles with a Cura namespace so that we can define our own settings in there.

filipgoc, 2016-11-14:
happy to hear the list is to be expanded a bit. 
(At least to add cool_fan_speed_max and cool_fan_speed_min (instead of just speed) etc. basically add categories of settings, instead of specific settings.)
(And I do agree with @MarcusWolschon that for many flexibles limiting max speeds make sense...)

ChrisTerBeke, 2017-11-23:
Do we want this? It might mean that something lower in the stack suddenly gets priority over something that's higher up in the stack?

nallath, 2017-11-27:
Preferably not. The data model is already complicated enough as it is.

ChrisTerBeke, 2017-11-27:
Agreed.

fieldOfView, 2017-11-27:
> It might mean that something lower in the stack suddenly gets priority over something that's higher up in the stack?

How would it do that? Please see (and preferably test) #2333, which implements the namespaced arbitrary cura setting concept mentioned by @Ghostkeeper. That enables setting speeds from the material profile. Any setting made there is neatly overridden by settings in higher containers (quality, quality changes and user). All fairly well behaved.

ChrisTerBeke, 2017-11-27:
@fieldOfView this behaviour is in the request of @MarcusWolschon ("This should be a part of the material profile to override the values from the printing profile"), which indeed goes against the whole stack setup so we probably won't do this. Loading Cura settings from an XML namespace is fine, but having them overrule the quality settings is not.

fieldOfView, 2017-11-27:
Right. It was getting confusing who was reacting to what.

ChrisTerBeke, 2017-11-27:
To be clear @MarcusWolschon: the request is to be able to set a maximum allowed printing speed in the material profile so that Cura gives a warning when a quality profile exceeds that value?

Ghostkeeper, 2017-12-11:
If we can set Cura's settings in material profiles we can override the maximum feedrate machine settings there.

ianpaschal, 2018-02-27:
Based on @nallath and @ChrisTerBeke's comments, I'm closing this.

closed 2018-02-27 by ianpaschal