Ultimaker/Cura #1143: feature suggestion: always show parent category when searching for an advanced setting

filipgoc, 2016-11-13:
This is what the advanced settings look like when scrolling through: 
[image]

But when one searches for a specific setting, the parent categories are hidden. 
This makes it useless for figuring out where specific setting lives!
It means I can enable any setting easily, but then have to still manually go through the UI settings to see where it actually is... 
[image]

**Solution: always display parent category in search.**

fieldOfView, 2016-11-13:
Yes, very much agreed. I'll see what I can do about this, unless team Cura or team Lulzbot beat me to it.

nallath, 2016-11-14:
This is already implemented for 2.4 :)

closed 2016-11-14 by nallath

fieldOfView, 2016-11-14:
By golly, you're right!