Ultimaker/Cura #1144: 2.3.1 Maximum Z Speed, and M203

darrelhunt, 2016-11-14:
Hi, new to Cura so pardon me if I'm doing something wrong.

I'm finding that when Cura creates the gcode, it's doing some math and it's placing M203 Z<Max Speed*60> into the gcode.
Which is logical it's converting to mm/min

And that'd work fine most of the time
Except, when using Marlin.
M203 is expected in mm/sec

My issue for example,
I like my Z speed to be 0.5 mm/sec (30 mm/min)

I put in 0.5 into the interface and it does the math, creating the mm/min:
M203 Z30
But Marlin expects that as the mm/sec and now it's running insanely fast and binds

In this case, entering 0.5 should just be passed straight through and M203 should be:
M203 Z0.5

Or

Change to the behavior to allow it to calculate to a higher decimal point

Currently it appears you can enter as many decimal places as you want, but it only calculates up to 3 decimal places, and even then there is a point between 0.010 and  0.020 and where it will either just calculate it as M203 Z1 or M203Z0

If it were allowed to calculate to a higher decimal place
You could enter something like 0.0084 and the math would make M203 Z<Something near 0.5>

Also Cura seems to place an M203 Z300 at the end of the gcode, right before the end gcode
Where is it told to do that?  And can it be stopped?

Also, first thing Cura does after the start gcode is retract, again where is it told to do that?  And can it be stopped?

Ghostkeeper, 2016-11-14:
Nice catch. We should be able to get this fixed.

I think the `M203 Z300` is to reset the maximum speed to the default, so that any subsequent print that does not specify the maximum speed will have the proper default maximum speed (unless you abort the print). I don't think there is an option to prevent it at the moment.

darrelhunt, 2016-11-14:
Glad I'm not going crazy.

About the Z Max Speed option in the GUI

After researching implementations of M203 in all of the RepRap firmwares.  Excluding Repetier as it doesn't use M203 for max feedrate, and I'm not including Ultimaker hardware in the RepRap classification.

It looks like all of them do M203 in mm/sec.  It seems that Cura shouldn't be multiplying it at all, rather than just passing the value through.  And I'd like to ask that it allows more granular in the GUI, say down to .01

Ghostkeeper, 2016-11-15:
You should be able to specify the speed down to 0.0001mm/s. It's a floating point field, and these accept values down to 4 digits after the radix.

darrelhunt, 2016-11-15:
True, the interface does allow.  But the resulting gcode gets rounded.  Looking at the code for Cura Engine, it looks like it's an integer.  Everything appears to get rounded there.  So the ability for it to output something like M203 Z0.5 doesn't exist.

darrelhunt, 2016-11-22:
Also a question.
Say a default install, Max Z Speed was never unhidden from advanced options, value never set.  M203 doesn't appear in gcode.

But if you unhide it and set a value, M203 appears in gcode

It wasn't performing the way I wanted so I though hiding it again would turn it of and M203 would no longer appear in gcode.

M203 remains, it seems like hiding it doesn't disable it.  And there's no option to disable it once you have enabled it.

Ghostkeeper, 2016-11-23:
I'd expect that iff the Max Z Speed setting is equal to the default, it wouldn't put the M203 commands in the g-code.

BagelOrb, 2016-11-29:
The M203 command is left out if it's set to zero...

BagelOrb, 2016-11-29:
M203 now always uses mm/s.
I've checked the firmware of Sprinter, Marlin and Repetier and all used mm/s.
I haven't checked any other firmware we support (BFB, MACH, MaketBot).

Other firmware types are not selectable in Cura.

closed 2016-11-29 by BagelOrb