Ultimaker/Cura #1145: ValueError: attempted relative import beyond top-level package

Zeroxus, 2016-11-14:
thank you for answering my last question,follow your tips,I fixed it. but another error happened!

```
`Error: Namespace packages not yet supported: Skipping package 'win32com.gen_py'
Traceback (most recent call last):
  File "setup_win32.py", line 95, in <module>
    options={"py2exe": {"skip_archive": False, "includes": includes}}
  File "C:\Users\Zeroxus\Anaconda3\lib\distutils\core.py", line 148, in setup
    dist.run_commands()
  File "C:\Users\Zeroxus\Anaconda3\lib\distutils\dist.py", line 955, in run_commands
    self.run_command(cmd)
  File "C:\Users\Zeroxus\Anaconda3\lib\distutils\dist.py", line 974, in run_command
    cmd_obj.run()
  File "C:\Users\Zeroxus\Anaconda3\lib\site-packages\py2exe\distutils_buildexe.py", line 188, in run
    self._run()
  File "C:\Users\Zeroxus\Anaconda3\lib\site-packages\py2exe\distutils_buildexe.py", line 267, in _run
    builder.analyze()
  File "C:\Users\Zeroxus\Anaconda3\lib\site-packages\py2exe\runtime.py", line 164, in analyze
    mf.import_hook(modname)
  File "C:\Users\Zeroxus\Anaconda3\lib\site-packages\py2exe\mf3.py", line 120, in import_hook
    module = self._gcd_import(name)
  File "C:\Users\Zeroxus\Anaconda3\lib\site-packages\py2exe\mf3.py", line 274, in _gcd_import
    return self._find_and_load(name)
  File "C:\Users\Zeroxus\Anaconda3\lib\site-packages\py2exe\mf3.py", line 357, in _find_and_load
    self._scan_code(module.__code__, module)
  File "C:\Users\Zeroxus\Anaconda3\lib\site-packages\py2exe\mf3.py", line 394, in _scan_code
    self.safe_import_hook(name, mod, fromlist, level)
  File "C:\Users\Zeroxus\Anaconda3\lib\site-packages\py2exe\mf3.py", line 138, in safe_import_hook
    self.import_hook(name, caller, fromlist, level)
  File "C:\Users\Zeroxus\Anaconda3\lib\site-packages\py2exe\mf3.py", line 123, in import_hook
    module = self._gcd_import(name, package, level)
  File "C:\Users\Zeroxus\Anaconda3\lib\site-packages\py2exe\mf3.py", line 274, in _gcd_import
    return self._find_and_load(name)
  File "C:\Users\Zeroxus\Anaconda3\lib\site-packages\py2exe\mf3.py", line 357, in _find_and_load
    self._scan_code(module.__code__, module)
  File "C:\Users\Zeroxus\Anaconda3\lib\site-packages\py2exe\mf3.py", line 394, in _scan_code
    self.safe_import_hook(name, mod, fromlist, level)
  File "C:\Users\Zeroxus\Anaconda3\lib\site-packages\py2exe\mf3.py", line 138, in safe_import_hook
    self.import_hook(name, caller, fromlist, level)
  File "C:\Users\Zeroxus\Anaconda3\lib\site-packages\py2exe\mf3.py", line 120, in import_hook
    module = self._gcd_import(name)
  File "C:\Users\Zeroxus\Anaconda3\lib\site-packages\py2exe\mf3.py", line 274, in _gcd_import
    return self._find_and_load(name)
  File "C:\Users\Zeroxus\Anaconda3\lib\site-packages\py2exe\mf3.py", line 357, in _find_and_load
    self._scan_code(module.__code__, module)
  File "C:\Users\Zeroxus\Anaconda3\lib\site-packages\py2exe\mf3.py", line 394, in _scan_code
    self.safe_import_hook(name, mod, fromlist, level)
  File "C:\Users\Zeroxus\Anaconda3\lib\site-packages\py2exe\mf3.py", line 138, in safe_import_hook
    self.import_hook(name, caller, fromlist, level)
  File "C:\Users\Zeroxus\Anaconda3\lib\site-packages\py2exe\mf3.py", line 120, in import_hook
    module = self._gcd_import(name)
  File "C:\Users\Zeroxus\Anaconda3\lib\site-packages\py2exe\mf3.py", line 274, in _gcd_import
    return self._find_and_load(name)
  File "C:\Users\Zeroxus\Anaconda3\lib\site-packages\py2exe\mf3.py", line 357, in _find_and_load
    self._scan_code(module.__code__, module)
  File "C:\Users\Zeroxus\Anaconda3\lib\site-packages\py2exe\mf3.py", line 394, in _scan_code
    self.safe_import_hook(name, mod, fromlist, level)
  File "C:\Users\Zeroxus\Anaconda3\lib\site-packages\py2exe\mf3.py", line 138, in safe_import_hook
    self.import_hook(name, caller, fromlist, level)
  File "C:\Users\Zeroxus\Anaconda3\lib\site-packages\py2exe\mf3.py", line 120, in import_hook
    module = self._gcd_import(name)
  File "C:\Users\Zeroxus\Anaconda3\lib\site-packages\py2exe\mf3.py", line 274, in _gcd_import
    return self._find_and_load(name)
  File "C:\Users\Zeroxus\Anaconda3\lib\site-packages\py2exe\mf3.py", line 357, in _find_and_load
    self._scan_code(module.__code__, module)
  File "C:\Users\Zeroxus\Anaconda3\lib\site-packages\py2exe\mf3.py", line 394, in _scan_code
    self.safe_import_hook(name, mod, fromlist, level)
  File "C:\Users\Zeroxus\Anaconda3\lib\site-packages\py2exe\mf3.py", line 138, in safe_import_hook
    self.import_hook(name, caller, fromlist, level)
  File "C:\Users\Zeroxus\Anaconda3\lib\site-packages\py2exe\mf3.py", line 120, in import_hook
[... 441 lines cut ...]
  File "C:\Users\Zeroxus\Anaconda3\lib\site-packages\py2exe\mf3.py", line 274, in _gcd_import
    return self._find_and_load(name)
  File "C:\Users\Zeroxus\Anaconda3\lib\site-packages\py2exe\mf3.py", line 357, in _find_and_load
    self._scan_code(module.__code__, module)
  File "C:\Users\Zeroxus\Anaconda3\lib\site-packages\py2exe\mf3.py", line 401, in _scan_code
    self._scan_code(c, mod)
  File "C:\Users\Zeroxus\Anaconda3\lib\site-packages\py2exe\mf3.py", line 394, in _scan_code
    self.safe_import_hook(name, mod, fromlist, level)
  File "C:\Users\Zeroxus\Anaconda3\lib\site-packages\py2exe\mf3.py", line 138, in safe_import_hook
    self.import_hook(name, caller, fromlist, level)
  File "C:\Users\Zeroxus\Anaconda3\lib\site-packages\py2exe\mf3.py", line 123, in import_hook
    module = self._gcd_import(name, package, level)
  File "C:\Users\Zeroxus\Anaconda3\lib\site-packages\py2exe\mf3.py", line 261, in _gcd_import
    name = self._resolve_name(name, package, level)
  File "C:\Users\Zeroxus\Anaconda3\lib\site-packages\py2exe\mf3.py", line 213, in _resolve_name
    raise ValueError('attempted relative import beyond top-level package')
ValueError: attempted relative import beyond top-level package
CMakeFiles\build_exe.dir\build.make:56: recipe for target 'build_exe' failed
mingw32-make[2]: *** [build_exe] Error 1
CMakeFiles\Makefile2:637: recipe for target 'CMakeFiles/build_exe.dir/all' failed
mingw32-make[1]: *** [CMakeFiles/build_exe.dir/all] Error 2
makefile:148: recipe for target 'all' failed
mingw32-make: *** [all] Error 2`

why py2exe would occured this error?How could I fix it? thank you!

Zeroxus, 2016-11-15:
this error has been solved
```
`Error: Namespace packages not yet supported: Skipping package 'win32com.gen_py'
```

but ValueError still occured
```
ValueError: attempted relative import beyond top-level package

closed 2016-11-15 by Zeroxus

Ghostkeeper, 2016-11-15:
Undoubtedly one of the longest tracebacks I've ever seen. Shame that it doesn't say where it attempts to import beyond the top-level package.

Zeroxus, 2016-11-16:
I am ashamed, I am a beginner.I apologize for my foolish behavior.at all events,thank you for your works.

Ghostkeeper, 2016-11-16:
Oh, don't be ashamed. That it doesn't say where it attempts to import beyond the top-level package is not your fault. It's py2exe's fault, if any.

Zeroxus, 2016-11-17:
Oh,stupid ! Cultural differences make me wrong to understand what you mean.Thanks for your reply.I think I should know more about your language habits.

Everything goes well with your work,thank you!