Ultimaker/Cura #1146: Cura-2.3.1.AppImage execution Segmentation fault (core dumped)

dasaki, 2016-11-15:
after receiving the update notice and downloading Cura-2.3.1.AppImage for linux, the application halts in the initialization process. Attached terminal dump.
[terminal_dump.txt](https://github.com/Ultimaker/Cura/files/591584/terminal_dump.txt)

Ghostkeeper, 2016-11-16:
Cleaned up by removing all occurrences of `...`:

    dasaki@dasaki-HP:~$ cd Descargas/
    dasaki@dasaki-HP:~/Descargas$ ./Cura-2.3.1.AppImage 
    Loaded plugin 3MFReader
    Loaded plugin AutoSave
    Loaded plugin ChangeLogPlugin
    2016-11-15 10:59:48,502 - INFO - /build/2.3/build/inst/lib/python3/dist-packages/UM/PluginRegistry.py (loadPlugin [63]): Loaded plugin ConsoleLogger
    2016-11-15 10:59:48,533 - INFO - /tmp/.mount_a1qqnv/usr/bin/plugins/plugins/CuraEngineBackend/CuraEngineBackend.py (__init__ [40]): Found CuraEngine at: /tmp/.mount_a1qqnv/usr/bin/CuraEngine
    2016-11-15 10:59:48,533 - INFO - /build/2.3/build/inst/lib/python3/dist-packages/UM/PluginRegistry.py (loadPlugin [63]): Loaded plugin CuraEngineBackend
    2016-11-15 10:59:48,538 - INFO - /build/2.3/build/inst/lib/python3/dist-packages/UM/PluginRegistry.py (loadPlugin [63]): Loaded plugin CuraProfileReader
    2016-11-15 10:59:48,543 - INFO - /build/2.3/build/inst/lib/python3/dist-packages/UM/PluginRegistry.py (loadPlugin [63]): Loaded plugin CuraProfileWriter
    2016-11-15 10:59:48,694 - INFO - /build/2.3/build/inst/lib/python3/dist-packages/UM/PluginRegistry.py (loadPlugin [63]): Plugin Doodle3D uses an incompatible API version, ignoring
    2016-11-15 10:59:48,699 - INFO - /build/2.3/build/inst/lib/python3/dist-packages/UM/PluginRegistry.py (loadPlugin [63]): Loaded plugin 3MFWriter
    2016-11-15 10:59:48,703 - INFO - /build/2.3/build/inst/lib/python3/dist-packages/UM/PluginRegistry.py (loadPlugin [63]): Loaded plugin OBJReader
    2016-11-15 10:59:48,706 - INFO - /build/2.3/build/inst/lib/python3/dist-packages/UM/PluginRegistry.py (loadPlugin [63]): Loaded plugin OBJWriter
    2016-11-15 10:59:48,736 - INFO - /build/2.3/build/inst/lib/python3/dist-packages/UM/PluginRegistry.py (loadPlugin [63]): Loaded plugin STLReader
    2016-11-15 10:59:48,741 - INFO - /build/2.3/build/inst/lib/python3/dist-packages/UM/PluginRegistry.py (loadPlugin [63]): Loaded plugin STLWriter
    2016-11-15 10:59:48,744 - INFO - /build/2.3/build/inst/lib/python3/dist-packages/UM/PluginRegistry.py (loadPlugin [63]): Loaded plugin FileLogger
    2016-11-15 10:59:48,748 - INFO - /build/2.3/build/inst/lib/python3/dist-packages/UM/PluginRegistry.py (loadPlugin [63]): Loaded plugin GCodeProfileReader
    2016-11-15 10:59:48,751 - INFO - /build/2.3/build/inst/lib/python3/dist-packages/UM/PluginRegistry.py (loadPlugin [63]): Loaded plugin GCodeWriter
    2016-11-15 10:59:48,759 - INFO - /build/2.3/build/inst/lib/python3/dist-packages/UM/PluginRegistry.py (loadPlugin [63]): Loaded plugin ImageReader
    2016-11-15 10:59:48,776 - INFO - /build/2.3/build/inst/lib/python3/dist-packages/UM/PluginRegistry.py (loadPlugin [63]): Loaded plugin LayerView
    2016-11-15 10:59:48,781 - INFO - /build/2.3/build/inst/lib/python3/dist-packages/UM/PluginRegistry.py (loadPlugin [63]): Loaded plugin LegacyProfileReader
    2016-11-15 10:59:48,787 - INFO - /build/2.3/build/inst/lib/python3/dist-packages/UM/PluginRegistry.py (loadPlugin [63]): Loaded plugin LocalFileOutputDevice
    2016-11-15 10:59:48,792 - INFO - /build/2.3/build/inst/lib/python3/dist-packages/UM/PluginRegistry.py (loadPlugin [63]): Loaded plugin MachineSettingsAction
    2016-11-15 10:59:48,799 - WARNING - /build/2.3/build/inst/lib/python3/dist-packages/UM/Preferences.py (getValue [70]): Tried to get the value of non-existing setting cura/active_mode.
    2016-11-15 10:59:48,799 - INFO - /build/2.3/build/inst/lib/python3/dist-packages/UM/PluginRegistry.py (loadPlugin [63]): Loaded plugin PerObjectSettingsTool
    2016-11-15 10:59:48,805 - INFO - /build/2.3/build/inst/lib/python3/dist-packages/UM/PluginRegistry.py (loadPlugin [63]): Loaded plugin PostProcessingPlugin
    2016-11-15 10:59:48,815 - INFO - /build/2.3/build/inst/lib/python3/dist-packages/UM/PluginRegistry.py (loadPlugin [63]): Loaded plugin RemovableDriveOutputDevice
    2016-11-15 10:59:48,833 - INFO - /build/2.3/build/inst/lib/python3/dist-packages/UM/PluginRegistry.py (loadPlugin [63]): Loaded plugin SliceInfoPlugin
    2016-11-15 10:59:48,838 - INFO - /build/2.3/build/inst/lib/python3/dist-packages/UM/PluginRegistry.py (loadPlugin [63]): Loaded plugin SolidView
    2016-11-15 10:59:48,845 - INFO - /build/2.3/build/inst/lib/python3/dist-packages/UM/PluginRegistry.py (loadPlugin [63]): Loaded plugin CameraTool
    2016-11-15 10:59:48,855 - INFO - /build/2.3/build/inst/lib/python3/dist-packages/UM/PluginRegistry.py (loadPlugin [63]): Loaded plugin MirrorTool
    2016-11-15 10:59:48,873 - INFO - /build/2.3/build/inst/lib/python3/dist-packages/UM/PluginRegistry.py (loadPlugin [63]): Loaded plugin RotateTool
    2016-11-15 10:59:48,889 - INFO - /build/2.3/build/inst/lib/python3/dist-packages/UM/PluginRegistry.py (loadPlugin [63]): Loaded plugin ScaleTool
    2016-11-15 10:59:48,893 - INFO - /build/2.3/build/inst/lib/python3/dist-packages/UM/PluginRegistry.py (loadPlugin [63]): Loaded plugin SelectionTool
    2016-11-15 10:59:48,901 - INFO - /build/2.3/build/inst/lib/python3/dist-packages/UM/PluginRegistry.py (loadPlugin [63]): Loaded plugin TranslateTool
    2016-11-15 10:59:48,910 - INFO - /build/2.3/build/inst/lib/python3/dist-packages/UM/PluginRegistry.py (loadPlugin [63]): Loaded plugin UltimakerMachineActions
    2016-11-15 10:59:48,995 - INFO - /build/2.3/build/inst/lib/python3/dist-packages/UM/PluginRegistry.py (loadPlugin [63]): Loaded plugin UM3NetworkPrinting
[... cut ...]
    2016-11-15 11:00:03,109 - WARNING - /build/2.3/build/inst/lib/python3/dist-packages/cura/Settings/ExtruderManager.py (getMachineExtruders [304]): Tried to get the extruder trains for machine Prusa i3, which doesn't exist.
    2016-11-15 11:00:03,109 - WARNING - /build/2.3/build/inst/lib/python3/dist-packages/cura/Settings/ExtruderManager.py (getMachineExtruders [304]): Tried to get the extruder trains for machine Prusa i3, which doesn't exist.
    Segmentation fault (core dumped)
    dasaki@dasaki-HP:~/Descargas$

awhiemstra, 2016-11-16:
Any chance you can run it through GDB? That should give us at least some idea of where to look. Note that you will need to do the following instead of just running the AppImage through gdb:

```
mount -o loop Cura-2.3.1.AppImage $mountpoint
cd $mountpoint/usr/bin
gdb cura
```

dasaki, 2016-11-16:
no idea about using gdb... 

dasaki@dasaki-HP:~/Descargas/curamount/usr/bin$ gdb ./cura
GNU gdb (Ubuntu 7.11.1-0ubuntu1~16.04) 7.11.1
Copyright (C) 2016 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.  Type "show copying"
and "show warranty" for details.
This GDB was configured as "x86_64-linux-gnu".
Type "show configuration" for configuration details.
Para las instrucciones de informe de errores, vea:
<http://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
<http://www.gnu.org/software/gdb/documentation/>.
For help, type "help".
Type "apropos word" to search for commands related to "word"...
Leyendo símbolos desde ./cura...(no se encontraron símbolos de depuración)hecho.
(gdb)

dasaki, 2016-11-16:
Note that previous version ( Cura-2.3.0.AppImage ) does run.

awhiemstra, 2016-11-16:
Oh, if you get to that point, you need to type "run" to actually start the program. Then once it crashes, type "bt" and paste the output you get here.

dasaki, 2016-11-16:
dasaki@dasaki-HP:~/Descargas/curamount/usr/bin$ gdb ./cura
GNU gdb (Ubuntu 7.11.1-0ubuntu1~16.04) 7.11.1
Copyright (C) 2016 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.  Type "show copying"
and "show warranty" for details.
This GDB was configured as "x86_64-linux-gnu".
Type "show configuration" for configuration details.
Para las instrucciones de informe de errores, vea:
<http://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
<http://www.gnu.org/software/gdb/documentation/>.
For help, type "help".
Type "apropos word" to search for commands related to "word"...
Leyendo símbolos desde ./cura...(no se encontraron símbolos de depuración)hecho.
(gdb) run
Starting program: /home/dasaki/Descargas/curamount/usr/bin/cura 
[Depuración de hilo usando libthread_db enabled]
Using host libthread_db library "/lib/x86_64-linux-gnu/libthread_db.so.1".
[Nuevo Thread 0x7ffff2b0c700 (LWP 7980)]
[Thread 0x7ffff2b0c700 (LWP 7980) terminado]
This application failed to start because it could not find or load the Qt platform plugin "xcb"
in "".

Reinstalling the application may fix this problem.

Thread 1 "cura" received signal SIGABRT, Aborted.
0x00007ffff6f0e428 in __GI_raise (sig=sig@entry=6)
    at ../sysdeps/unix/sysv/linux/raise.c:54
54	../sysdeps/unix/sysv/linux/raise.c: No existe el archivo o el directorio.
(gdb) bt
#0  0x00007ffff6f0e428 in __GI_raise (sig=sig@entry=6)
    at ../sysdeps/unix/sysv/linux/raise.c:54
#1  0x00007ffff6f1002a in __GI_abort () at abort.c:89
#2  0x00007fffe987bcbe in QMessageLogger::fatal(char const*, ...) const ()
   from /home/dasaki/Descargas/curamount/usr/bin/libQt5Core.so.5
#3  0x00007fffe4fcbf3c in QGuiApplicationPrivate::createPlatformIntegration()
    () from /home/dasaki/Descargas/curamount/usr/bin/libQt5Gui.so.5
#4  0x00007fffe4fcbf8d in QGuiApplicationPrivate::createEventDispatcher() ()
   from /home/dasaki/Descargas/curamount/usr/bin/libQt5Gui.so.5
#5  0x00007fffe9a5bcdd in QCoreApplicationPrivate::init() ()
   from /home/dasaki/Descargas/curamount/usr/bin/libQt5Core.so.5
#6  0x00007fffe4fccc4c in QGuiApplicationPrivate::init() ()
   from /home/dasaki/Descargas/curamount/usr/bin/libQt5Gui.so.5
#7  0x00007fffe3a77859 in QApplicationPrivate::init() ()
   from /home/dasaki/Descargas/curamount/usr/bin/libQt5Widgets.so.5
#8  0x00007fffe4217249 in sipQApplication::sipQApplication(int&, char**, int)
    ()
   from /home/dasaki/Descargas/curamount/usr/bin/lib/python3.5/PyQt5.QtWidgets.so
#9  0x00007fffe4217315 in init_type_QApplication ()
   from /home/dasaki/Descargas/curamount/usr/bin/lib/python3.5/PyQt5.QtWidgets.so
#10 0x00007fffe8e4ce93 in ?? ()
---Type <return> to continue, or q <return> to quit---
   from /home/dasaki/Descargas/curamount/usr/bin/lib/python3.5/sip.so
#11 0x000000000047441c in ?? ()
#12 0x00000000004236fa in PyObject_Call ()
#13 0x00000000004d96d5 in PyEval_EvalFrameEx ()
#14 0x00000000004e1e4d in ?? ()
#15 0x00000000004de7f1 in PyEval_EvalFrameEx ()
#16 0x00000000004e1e4d in ?? ()
#17 0x00000000004e1f23 in PyEval_EvalCodeEx ()
#18 0x000000000058f075 in ?? ()
#19 0x00000000004236fa in PyObject_Call ()
#20 0x0000000000581f6d in ?? ()
#21 0x00000000004236fa in PyObject_Call ()
#22 0x000000000047bff9 in ?? ()
#23 0x0000000000478f06 in ?? ()
#24 0x00000000004236fa in PyObject_Call ()
#25 0x00000000004db313 in PyEval_EvalFrameEx ()
#26 0x00000000004e003c in PyEval_EvalFrameEx ()
#27 0x00000000004e1e4d in ?? ()
#28 0x00000000004e1f23 in PyEval_EvalCodeEx ()
#29 0x00000000004e1f4b in PyEval_EvalCode ()
#30 0x00000000004d5ffd in ?? ()
#31 0x00000000005940c9 in PyCFunction_Call ()
#32 0x00000000004e0943 in PyEval_EvalFrameEx ()
---Type <return> to continue, or q <return> to quit---
#33 0x00000000004e1e4d in ?? ()
#34 0x00000000004e1f23 in PyEval_EvalCodeEx ()
#35 0x00000000004e1f4b in PyEval_EvalCode ()
#36 0x000000000041c919 in main ()
(gdb)

Ghostkeeper, 2016-11-16:
That helps! It can't load xcb for Qt. Are we including libQt5DBus in the AppImage?

dasaki, 2016-11-17:
Appart, it seems that with the appimage (2.3.0) I can't edit the printing profiles as the text boxes aren't editable or do not respond to the keyboard.
[image]

awhiemstra, 2016-11-17:
That's a bug that was fixed for the 2.3.1 AppImage. For the 2.3.0 you can workaround it by launching it with `QT_XKB_CONFIG_ROOT=/usr/share/X11/xkb ./Cura-2.3.0.AppImage`.

dasaki, 2016-11-20:
ok, thanks!

dasaki, 2016-11-20:
I tried T_XKB_CONFIG_ROOT=/usr/share/X11/xkb ./Cura-2.3.0.AppImage, but it still doesn't let edit the parameters.

awhiemstra, 2016-11-21:
QT_XKB_CONFIG_ROOT, not T_ . Unless that was a copy paste error, in which case I do not really know what is happening.

dasaki, 2016-11-21:
Ok, it works, thanks!
It happened that I just tested the editing by hitting an arbitrary key in a text box that only accepts numbers :)

DanielJoyce, 2017-01-19:
I get a crash on KUbuntu when I try and set the printer. No unity, just KDE5.

Ghostkeeper, 2017-01-22:
What do you mean by "set the printer"? Do you mean adding a printer (select a printer model from the menu for the first time, etc.) or do you mean selecting a printer from the drop-down menu?

DanielJoyce, 2017-01-22:
When first launched and running the setup wizard asks you to choose a
printer. Selecting a print by clicking on a radio button causes an
immediate​ crash.

Skipping the printer select step and trying to configure it later causes
immediate crash as soon as you click on the radio button

<https://github.com/Ultimaker/Cura/issues/1146#issuecomment-274361695>, or mute
the thread

Daniel Joyce

The meek shall inherit the Earth, for the brave will be among the stars.

Ghostkeeper, 2017-01-23:
That seems to be indeed the issue of https://github.com/Ultimaker/Cura/issues/1219 then, Daniel. You also mentioned it there.

DanielJoyce, 2017-01-23:
Shoot sorry. Somehow I am on both and crossposted.

nallath, 2017-07-27:
I can't reproduce this, and since we're 3 versions ahead, this is probably fixed a long time ago.

closed 2017-07-27 by nallath

Ghostkeeper, 2017-08-10:
In fact, it was fixed in 2.3.1 as Awhiemstra said.