Ultimaker/Cura #1148: Cura 2.3.0 slicing weirdness

Klipi, 2016-11-15:
Hello,

I'm trying to generate .stl files and eventually 3D-print them. I'm running into some issues with the slicing. Here's an image of the layers after slicing with default settings:

[image]

Here's the same model in Cura after opening it in Meshlab, importing with the Unify Duplicated Vertices setting and exporting again:

[image]

As you can see, the hair and ear get cut in the first model. I'm wondering what's the cause for this, since pretty much all 3D-programs render both .stl-files identically?

Here's both .stl files (avatar_original.stl is the one with the issue and avatar_unified.stl is the one that works):

[avatars.zip](https://github.com/Ultimaker/Cura/files/591905/avatars.zip)

BerndJM, 2016-11-15:
There are some errors in your original model - if you have a look to the model in X-ray view you'll see some red parts - this are the critical parts which confuse the slicer. After your repair with Meshlab this errors are nearly gone and the slicer can do a better job.
Rendering a file for 3D-view and slicing it for printing are two total different assignment of tasks ...

Ghostkeeper, 2016-11-16:
Note that the new model still has extra faces in the volume. Just a bit though, at one of the bangs.

Not to mention that the model could use a hair cut :wink:

In any case, it doesn't seem to be an error on Cura's side. If there are extra walls in the middle of the model or the model is not watertight, it's ambiguous what the inside of the mesh is. It finds a wall there and thinks that is the end of the volume, but there happens to be another wall just a few millimetres ahead.

closed 2016-11-16 by Ghostkeeper