Ultimaker/Cura #1154: Sorting slicing profiles variables on export?

filipgoc, 2016-11-17:
What is the order in which the variables get written into profiles when we export slicing settings? 
It seems to be all over the place. 

If we do some sort of sorting (categorical?!), it would help keep profiles in check and developing them.

Example: 
retraction all over the place, speeds all over the place, cooling all over the place, extrusion all over the place... 
```
[values]
retraction_retract_speed = 70
skin_no_small_gaps_heuristic = False
coasting_enable = True
cool_min_layer_time = 4
top_thickness = 0.8
adhesion_type = skirt
skirt_brim_minimal_length = 40
retraction_speed = 70
speed_wall_x = 35
infill_before_walls = False
retraction_amount = 1.3
speed_travel_layer_0 = 80
cool_fan_speed_max = 100
material_flow = 90
bottom_thickness = 0.6
retraction_combing = noskin
meshfix_union_all = False
speed_layer_0 = 20
retraction_prime_speed = 30
infill_pattern = zigzag
infill_overlap = 15
cool_fan_full_at_height = 0.4
line_width = 0.4
wall_thickness = 0.8
coasting_speed = 95
skirt_brim_speed = 20
cool_fan_speed_min = 50
retraction_min_travel = 1.2
speed_travel = 120
speed_topbottom = 25
cool_min_layer_time_fan_speed_max = 5
layer_height_0 = 0.3
speed_print = 45
skirt_line_count = 3
retraction_hop_enabled = 0.1
infill_sparse_density = 20
speed_slowdown_layers = 1
cool_min_speed = 10
speed_wall = 25
layer_height = 0.3
infill_line_width = 0.6
```

filipgoc, 2016-11-17:
I'm just doing quick alphabetical sort now, and it's not that bad either. (good for diffs, not great for first reads)

fieldOfView, 2016-11-17:
The order is currently random, because Python dicts don't have an order

filipgoc, 2016-11-17:
Ok. thanks.

nickthetait, 2016-11-17:
I think there is incredible value compared to (what seems like) a small amount of effort for this feature. :+1:

nallath, 2016-11-18:
I think our materials team would also want this, because right now any form of diff checking fails horribly.

Ghostkeeper, 2016-11-18:
Trivial addition: https://github.com/Ultimaker/Uranium/commit/2a5c6e7914160fc40ab8f8f5462f7016b362993c

closed 2016-11-18 by Ghostkeeper