Ultimaker/Cura #1158: Cura ppa UM3 connect missing

peteruithoven, 2016-11-18:
I'm not sure whether this is the right place for this issue, but I'm having an issue with the Cura version on the ppa. For some reason the Ultimaker 3 connect buttons are missing. @thopiekar is this something you can help with? 

Cura 2.3.1 AppImage:
[image]

Cura 2.3.1 PPA:
[image]

Let me know if there is something I can check.

thopiekar, 2016-11-18:
Hey @peteruithoven ,

yes, this is the right place. Also my own repository called "Cura-packaging" would be fine, but as long as you mention me here, I get directly notified.

You probably missed to add needed plugin for the UM3
Here is the list of extra plugins, which are available in my stable PPA:
```
cura-extra-plugin-octoprint Cura plugin - OctoPrint
cura-extra-plugin-postprocessing Cura plugin - PostProcessing
cura-extra-plugin-um3networkprinting Cura plugin - UM3NetworkPrinting
cura-extra-plugins-all Cura plugin meta-package
```
Make sure you've installed `cura-extra-plugins-all` or `cura-extra-plugin-um3networkprinting`.

Finally, have you used any instruction to install Cura via PPA?
I guess I should mention the extra plugins in the thread in the Ultimaker forums :thinking:

peteruithoven, 2016-11-18:
Awesome. After installing those the connect button appeared right away. 

Ah, sorry never found those instructions. Only read the instructions on:
https://launchpad.net/~thopiekar/+archive/ubuntu/cura

thopiekar, 2016-11-18:
No problem. Wish you a lot of fun!

Ah, these are outdated. Thank you for the hint.
I'm at the moment working on the PPA. So as it seems to work for you at the moment, please don't do any updates to Cura at the moment.
Will give you feedback (or simply close the issue here) when everything is save again :wink:

thopiekar, 2016-11-18:
Stable PPA is now back again and completely built on 2.3.1, as Ultimaker does.

closed 2016-11-18 by thopiekar