Ultimaker/Cura #1159: Cura not inserting temperatures in startup template gcode?

climbhigh09, 2016-11-19:
Here's what Cura is generating:

> ;FLAVOR:Repetier
> ;TIME:4099
> ;Generated with Cura_SteamEngine 2.3.0
> M190 S45
> M104 S240
> M109 S240
> G28
> G0 Z0 X0 Y90 F3500
> M104 S<TEMP>
> M190 S<BEDTEMP>
> M109 S<TEMP>

The S<TEMP> values cause my Rostock to set targets to 0 and sit there with no forward action.  Seems like the first set of pre-heat temps should be in these locations (want my printhead to warm up on the bed).

Thoughts?

fieldOfView, 2016-11-19:
Have you set a custom start gcode in the machine settings? If so, could to post it here?

climbhigh09, 2016-11-19:
There it is, I thought I'd checked that:

G28
G0 Z0 X0 Y90 F3500
M104 S<TEMP>
M190 S<BEDTEMP>
M109 S<TEMP>
G28

Obviously (embarrassingly) the source of my problem and likely a hold-over from another slicer program.  However, I'd like to achieve this with Cura- the pause at 0,90,0 while the extruder heats- is there a similar variable to the  <temp> and <bedtemp> that Cura can populate on a print-by-print basis?

fieldOfView, 2016-11-20:
Try this:
```
G28
G0 Z0 X0 Y90 F3500
M104 S{material_print_temperature}
M190 S{material_bed_temperature}
M109 S{material_print_temperature}
G28
```

climbhigh09, 2016-11-20:
All better, many thanks.

closed 2016-11-20 by climbhigh09