Ultimaker/Cura #1160: Cura 2.3.1 failing to connect to smoothieware based controllesr and cause the printer to hang.

eccentrichead, 2016-11-19:
Issue description: 
My printer is a kossel mini build which has the MKS sbase 32 bit controller board which is smoothieboard clone. I have recently installed cura 2.3.1 version from  "ppa:thopiekar/cura" on my Ubuntu 16 laptop.  What I see is now, when I plug the printer to my USB port and launch the cura GUI, the controller board tends to go in a hang like state and the cura fails to connect to the printer. The "cura.logs" shows up following message while I try connecting the printer:
~~~~~~
2016-11-19 15:34:28,718 - DEBUG - .../UM/Backend/Backend.py (_logSocketState [165]): Socket state changed to Connected
2016-11-19 15:34:28,718 - DEBUG - .../UM/Backend/Backend.py (_onSocketStateChanged [155]): Backend connected on port 49675
2016-11-19 15:34:37,543 - DEBUG - /usr/lib/cura/plugins/AutoSave/AutoSave.py (_onTimeout [46]): Autosaving preferences, instances and profiles
2016-11-19 15:35:20,744 - DEBUG - /usr/lib/cura/plugins/USBPrinting/USBPrinterOutputDevice.py (_connect [298]): Attempting to connect to /dev/ttyACM0
2016-11-19 15:35:20,745 - WARNING - .../UM/OutputDevice/OutputDeviceManager.py (removeOutputDevice [123]): Could not find output device with id /dev/ttyACM0 to remove
2016-11-19 15:35:22,365 - INFO - /usr/lib/cura/plugins/USBPrinting/USBPrinterOutputDevice.py (_connect [298]): Could not establish connection on /dev/ttyACM0: 'Timeout'. Device is not arduino based.
2016-11-19 15:35:22,365 - DEBUG - /usr/lib/cura/plugins/USBPrinting/USBPrinterOutputDevice.py (_connect [298]): Attempting to connect to printer with serial /dev/ttyACM0 on baud rate 115200
2016-11-19 15:36:06,421 - ERROR - /usr/lib/cura/plugins/USBPrinting/USBPrinterOutputDevice.py (_sendCommand [389]): Unexpected error while writing serial port write failed: [Errno 5] Input/output error
2016-11-19 15:36:06,421 - DEBUG - /usr/lib/cura/plugins/USBPrinting/USBPrinterOutputDevice.py (close [363]): Closing the USB printer connection.
2016-11-19 15:36:06,421 - DEBUG - /usr/lib/cura/plugins/USBPrinting/USBPrinterOutputDevice.py (close [363]): PrinterConnection.close: cannot join current thread (expected)
2016-11-19 15:36:06,422 - WARNING - .../UM/OutputDevice/OutputDeviceManager.py (removeOutputDevice [123]): Could not find output device with id /dev/ttyACM0 to remove
2016-11-19 15:36:07,485 - DEBUG - /usr/lib/cura/plugins/USBPrinting/USBPrinterOutputDevice.py (_connect [298]): Attempting to connect to printer with serial /dev/ttyACM0 on baud rate 250000
2016-11-19 15:36:07,485 - DEBUG - /usr/lib/cura/plugins/USBPrinting/USBPrinterOutputDevice.py (_connect [298]): Could not open port /dev/ttyACM0
2016-11-19 15:36:07,485 - DEBUG - /usr/lib/cura/plugins/USBPrinting/USBPrinterOutputDevice.py (_connect [298]): Attempting to connect to printer with serial /dev/ttyACM0 on baud rate 230400
2016-11-19 15:36:07,485 - DEBUG - /usr/lib/cura/plugins/USBPrinting/USBPrinterOutputDevice.py (_connect [298]): Could not open port /dev/ttyACM0
2016-11-19 15:36:07,486 - DEBUG - /usr/lib/cura/plugins/USBPrinting/USBPrinterOutputDevice.py (_connect [298]): Attempting to connect to printer with serial /dev/ttyACM0 on baud rate 57600
2016-11-19 15:36:07,486 - DEBUG - /usr/lib/cura/plugins/USBPrinting/USBPrinterOutputDevice.py (_connect [298]): Could not open port /dev/ttyACM0
2016-11-19 15:36:07,486 - DEBUG - /usr/lib/cura/plugins/USBPrinting/USBPrinterOutputDevice.py (_connect [298]): Attempting to connect to printer with serial /dev/ttyACM0 on baud rate 38400
2016-11-19 15:36:07,486 - DEBUG - /usr/lib/cura/plugins/USBPrinting/USBPrinterOutputDevice.py (_connect [298]): Could not open port /dev/ttyACM0
2016-11-19 15:36:07,487 - DEBUG - /usr/lib/cura/plugins/USBPrinting/USBPrinterOutputDevice.py (_connect [298]): Attempting to connect to printer with serial /dev/ttyACM0 on baud rate 19200
2016-11-19 15:36:07,487 - DEBUG - /usr/lib/cura/plugins/USBPrinting/USBPrinterOutputDevice.py (_connect [298]): Could not open port /dev/ttyACM:
2016-11-19 15:36:07,488 - DEBUG - /usr/lib/cura/plugins/USBPrinting/USBPrinterOutputDevice.py (_connect [298]): Attempting to connect to printer with serial /dev/ttyACM0 on baud rate 9600
2016-11-19 15:36:07,489 - DEBUG - /usr/lib/cura/plugins/USBPrinting/USBPrinterOutputDevice.py (_connect [298]): Could not open port /dev/ttyACM0
2016-11-19 15:36:07,490 - ERROR - /usr/lib/cura/plugins/USBPrinting/USBPrinterOutputDevice.py (_connect [298]): Baud rate detection for /dev/ttyACM0 failed
2016-11-19 15:36:07,491 - DEBUG - /usr/lib/cura/plugins/USBPrinting/USBPrinterOutputDevice.py (close [363]): Closing the USB printer connection.
2016-11-19 15:36:10,787 - DEBUG - /usr/lib/cura/plugins/USBPrinting/USBPrinterOutputDevice.py (close [363]): Closing the USB printer connection.
2016-11-19 15:36:25,805 - DEBUG - /usr/lib/cura/plugins/USBPrinting/USBPrinterOutputDevice.py (_connect [298]): Attempting to connect to /dev/ttyACM0
2016-11-19 15:36:25,806 - WARNING - .../UM/OutputDevice/OutputDeviceManager.py (removeOutputDevice [123]): Could not find output device with id /dev/ttyACM0 to remove
2016-11-19 15:36:30,496 - INFO - /usr/lib/cura/plugins/USBPrinting/USBPrinterOutputDevice.py (_connect [298]): Could not establish connection on /dev/ttyACM0: 'Timeout'. Device is not arduino based.
2016-11-19 15:36:30,496 - DEBUG - /usr/lib/cura/plugins/USBPrinting/USBPrinterOutputDevice.py (_connect [298]): Attempting to connect to printer with serial /dev/ttyACM0 on baud rate 115200
2016-11-19 15:37:13,437 - ERROR - /usr/lib/cura/plugins/USBPrinting/USBPrinterOutputDevice.py (_sendCommand [389]): Unexpected error while writing serial port write failed: [Errno 5] Input/output error
2016-11-19 15:37:13,437 - DEBUG - /usr/lib/cura/plugins/USBPrinting/USBPrinterOutputDevice.py (close [363]): Closing the USB printer connection.
2016-11-19 15:37:13,437 - DEBUG - /usr/lib/cura/plugins/USBPrinting/USBPrinterOutputDevice.py (close [363]): PrinterConnection.close: cannot join current thread (expected)
2016-11-19 15:37:13,438 - WARNING - .../UM/OutputDevice/OutputDeviceManager.py (removeOutputDevice [123]): Could not find output device with id /dev/ttyACM0 to remove
~~~~~~

At the sametime after resetting the printer, I can connect to it from all other apps like slic3r, repetier or pronterface. Smoothie is configured to connect with baudrate 115200. I had tried other baudrates as well, the issue continues to present.

I had never faced such a problem while using cura 15  version, The connections was happening smoothly with no issue whatsoever.

Affected Environment details:
* DELL 15 latop running with  Ubuntu16.04 which has the latest patches applied.
* MKS SBASE 32-bit printer controller with latest smoothieware edge firmware version.
* Cura 2.3.X version from"ppa:thopiekar/cura" .

Ghostkeeper, 2016-11-21:
Weird. What poison could we be sending through that USB cable to cause a smoothie to hang?

It seems like the rest of the errors we see here are because the device doesn't respond any more.

The code by which we connect hasn't changed really since June 2015: https://github.com/Ultimaker/Cura/blame/master/plugins/USBPrinting/USBPrinterOutputDevice.py#L313
Except a bit of logging and documentation...

eccentrichead, 2016-11-21:
Not sure what happens there but as soon as i launch cura 2.3 the board hangs up. I have presently reinstalled cura 15 which works without any issues. I have the complete logs if you need. If you need some more test result i can reinstall and get it for reference.

ianpaschal, 2018-02-21:
No more work will be done on versions 2.x. If this is still occurring in versions 3.2 or newer, we can re-open this issue.

closed 2018-02-21 by ianpaschal

Ghostkeeper, 2018-03-13:
Some log of the printer itself might help. It's a firmware bug that the printer hangs, but a log from there might give some indication of why it would hang.