Ultimaker/Cura #1161: Post Processing not executed when "Print with Octoprint"

bbaumg, 2016-11-19:
I have observed that when using the "Print with Octoprint" to send the print job straight to my printer, the post-processing actions are not applied.  For example, when I added a "pause at height" and send the print job to the printer with octoprint and also save to file, the file sent to octoprint is completely missing the post-processing commands.

This code was in the file saved by cura to disk:
;added code by post processing
;script: PauseAtHeight.py
;current z: 2.100000 
M83
G1 Z3.100000 F300
G1 X0.000000 Y0.000000 F9000
G1 Z15 F300
M84 E0
M0 ;Do the actual pause

The above code was completely missing from the file I downloaded from the octoprint server after using cura to send it Octoprint.

thopiekar, 2016-11-19:
Probably @fieldOfView can give an answer to this 😉

fieldOfView, 2016-11-20:
I don't know, I would have to debug this. @bbaumg, could you create an issue at the OctoPrintPlugin repository so I don't forget? 
https://github.com/fieldOfView/OctoPrintPlugin/issues

fieldOfView, 2016-11-29:
I have fixed this in the OctoPrint plugin:
https://github.com/fieldOfView/OctoPrintPlugin/commit/18a806adb3ff8a3e6799b768488cb7944758b514
Same fix needs to be applied in UM3NetworkPrinter plugin:
https://github.com/Ultimaker/Cura/pull/1196

closed 2016-11-30 by nallath

bbaumg, 2016-12-04:
I have tested and it works great!
Thanks,

fieldOfView, 2016-12-05:
Thanks for confirming