Ultimaker/Cura #1162: BUG: Cura 2.3.1 - Surface Mode - Both

PDBeal, 2016-11-19:
While trying to print the Fly-Wing2.STL file scaled to 2540% (primarily to get it from inches to mm) from http://www.thingiverse.com/thing:1852573, Cura 2.3.1 doesn't recognize the Surface Mode set to Both.

If you select the option Surface, it slices the lines on the wings and only generates G-code for the outer surface.  If you selection the option Normal, it ignores the lines on the wings and only prints the wings without the detail.  If you selection the option Both, I was expecting it to slice the wings and the surface lines but it appears it only slices the wings making it no different than the option Normal.

I've tried this on Mac OS and Windows in both Cura 2.3 and Cura 2.3.1 with the same result.

Ghostkeeper, 2016-11-21:
This is a misunderstanding of the Mesh Surface Mode setting.

In Normal mode, what happens is that it tries to fill the wings but fails for the most part because it can't fit a border inside the wing. The result is that the print ends up for the most part empty, with a few bits of plastic being printed in mid-air.
In Surface mode it just takes all surfaces and traces them with a line. This technique is very dumb but it may be what you need if the model is not water tight (this model is water tight though).
When using Both, it takes the volumes that it can using Normal mode, and all the extra triangles that are not part of the water tight mesh are traced with Surface mode. So in this case, it is indeed equal to Normal mode since there are no extra triangles leftover after all the normal volumes are "filled".

In any case, I'd certainly print those wings laying down instead of on the tips of the wings!

closed 2016-11-21 by Ghostkeeper

PDBeal, 2016-11-21:
So basically, this is an STL file that Cura is unable to process and print.  I have no issue with this file in Slic3r and it pulls the details properly and produces the correct g-code to produce the part.

Ghostkeeper, 2016-11-22:
Cura can process the STL file. Printing it though requires you to adjust one of these:
* Rotate the part 90 degrees such that it lays down flat (highly advisable).
* Use a smaller line width such that the lines actually fit inside the mesh. Cura removes details that are too small to print with your line width.
* Expand the part slightly with the Horizontal Expansion setting such that the lines fit inside the mesh.