Ultimaker/Cura #1163: Cura 2.3.1 Wrong number of Layers

pelikan39, 2016-11-20:
The attached file  showes 4 layers  instead of 10  (hight  0.4 / layer hight 0.04 =10) Cura 15.04.4 works fine.
Is this a err
[image]
or or did  I made a configuration mistake?

fieldOfView, 2016-11-20:
There's a setting called "Initial Layer Height" that is currently not visible in your sidebar. It's default value is 0.3.  ```0.3 + 3 * 0.04 = 0.42```, so you get 4 layers.

Click the cog-wheel next to Quality to make the Initial Layer Height setting visible. Note that you may get first layer issues with an initial layer height this small.

closed 2016-11-21 by Ghostkeeper

pelikan39, 2016-11-22:
Thank you very much for your awnser, I followed your advise, now it works fine.
Yes initial layer hight this small is difficult, but if the bed leveling is ok it works.