Ultimaker/Cura #1165: No "First layer line width multiplier"

Montiey, 2016-11-21:
A couple of things:
Maybe there is a way around this, but I haven't found it in the past couple of months.
For most repraps and lower-end printers, or even higher end models and finicky filaments, you pretty much always need a fat first layer to get a good adhesion. Problem is, you can change the width of the bottom infill (solid) independently, but not the line width of the inner or outer walls. This leads to an issue where if you raise them to get the first layer down, the lines will be too big to print some small walls later on.

What I suggest is an isolated multiplier (or set of multipliers) for all lines on the first layer nothing else. This would let us poor folk finally get a reliable layer that won't "string up" and ruin everything, but importantly,  still make printing small walls a reality.

:)

Ghostkeeper, 2016-11-21:
You can still increase the layer height for the first layer and then adjust the bed higher.
Increasing the line width for the first layer also makes the lines further apart, so that there is in theory no net difference. In practice, if the nozzle is not wide enough, it creates holes in the first layer which make it stick worse.

Ghostkeeper, 2016-11-21:
What would maybe be handy is a flow setting for the first layer. Increasing the first layer height has two disadvantages: Elephant feet (which is more or less what you want since it increases the surface to the build plate) and dimensional inaccuracy. It "eats" a bit of the model from the bottom side. The second problem can be resolved if you'd increase the flow in just the first layer instead of increasing the layer height.

noisemaker00, 2016-11-22:
Same problem: I switched from Slic3r to Cura and the first problem I notice was objects do not stick on bed, even if heated to 90°C. 
With Slic3r no problem with this: every printi I made (with PLA or PETG) sticked perfectly to the bed simply setting the extrusion width a bit higher than 100%.
Please add this feature.

weakset, 2017-01-30:
1st layer flow setting would actually work better than 1st layer width. (If you have to choose one.) Flow only compensates uneven bed by filling low areas and over extruding in high areas, but wont ruin thin walls. You only have to be more carefull not using too high value.

I did testing with hacked TweakAtZ plugin and got as good results as with Cura 15.04.6 and "Initial layer line width" setting. By default the plugin won't work with layer 0. But using plugin is annoying when you need it with almost every raftless print.

whosawhatsis, 2017-04-28:
I print with the first layer extrusion width set to ~200%. This sticks reliably and makes the printer much less sensitive to platform adjustment. Just increasing the layer height for the first layer doesn't work because for good contact you want a high aspect ratio AND high extrusion, which you can only do by making the lines wider. It's like the first layer of a raft, but it doesn't need to be removed after printing because it's just the first layer of the object.

I'm still using the old Cura primarily, and occasionally Slic3r, MatterControl, or S3D, all of which have this feature. Slicers that don't have it are, in my opinion, unusable.

DDDirk, 2017-04-29:
The first layer not sticking well enough can be caused by underextrusion. One solution is to lower nominal extrusion, because the higher that is, the more underextrusion you get. On an UM2 underextrusion already starts above 1 mm³/s and at 4 mm³/s there's already about 10% underextrusion. The more powerful extruder on an UM2+ should improve this, but I don't have figures for that. (Also, it will depend on the filament.)

Fiddling with the nominal extrusion value is one solution, but how much is needed depends on the printer and other factors. Setting it too high may affect dimensions, especially with a high layerheight.

A simple solution is to print at a very low speed. Since it's only the first layer, that will not increase print time too much.
The nominal extrusion value is
road width * layer height * speed
The initial layerheight should not be too low (partly because the glass is never perfectly flat). So let's say 0.3 mm. And if the nozzle size / road width is 0.4 mm, then a print speed of 10 mm/s will give 0.3 * 0.4 * 10 = 1.2 mm³/s. To be absolutely sure, you might even go lower (but be careful with that on other layers, because the nozzle may then melt the previous layer).

And it doesn't help with filament that doesn't have a constant width, but even then it will improve things.
And of course, it depends on the print bed and any adhesive you might put on it.

Oh, and a low speed will also reduce the chance that roads curl up when the print head takes a (sharp) turn.

timsamart, 2017-07-01:
I have written a very crude python script which offsets all z values in a gcode by a certain amount (0.3mm). for me that was enough to get going. if someone needs it:
https://github.com/timsamart/gcodeZoffset

14bitVoid, 2017-07-09:
2 months ago I made this in cura engine, because it was one of the top issues in github then and I thought I could do it. But actually it was pretty hard so I didn't finish it, but now I implemented the rest: https://github.com/Ultimaker/CuraEngine/pull/557

Please check it out!

PR #2088 (2017-07-11): First layer line width Multiplier
Multiplies the line width on just the first layer.

for a description: https://github.com/Ultimaker/CuraEngine/pull/557
Fixes #1165
closed 2017-07-11 by Ghostkeeper