Ultimaker/Cura #1166: Brim bug

thopiekar, 2016-11-21:
> The "Brim only on the Outside" function is not working in the best way.
> 
> Has you can see in the imagens, after selected that option, brim still appears in the inside.
> 
> Brim only on the Outside unchecked

[image]

Resource: https://ultimaker.com/en/community/33735-cura-231-brim-bug?page=1

Ghostkeeper, 2016-11-22:
The screenshot only shows expected behaviour: A brim is created around every object, even around objects that are enclosed by other objects.

On the forum you find a close-up of a "B" that shows that the brim also comes from the outside if there is another object inside. I don't remember exactly if that is expected behaviour but it looks strange.

peteruithoven, 2016-12-06:
Maybe it's slightly confusing on what one object is? I created for an as simple as possible example of one stl with 2 shapes:
[image]
[target.stl.tar.gz](https://github.com/Ultimaker/Cura/files/634056/target.stl.tar.gz)

Maybe there needs to be more options? 
- Brim only on outside
- Brim only on outside of shapes

Ghostkeeper, 2016-12-07:
The behaviour is slightly different. It doesn't depend on the number of shapes.

The catch is that it only removes the brim from holes that are empty. The thought was that if a hole is not empty, adding an inside brim would let it attach better to the outside brim of whatever is inside, I think.

maksdampf, 2018-02-20:
I can confirm that "Brim only on outside" is currently not working in 3.2.1 on macOS.

There is no difference in slicing output with the feature enabled or disabled. 
[image]

BagelOrb, 2018-02-20:
> @BagelOrb <https://github.com/bagelorb> Can you reassign the label
> "release:current" to the issue please?

It seems to be working as intended.

1. The brim is on the outside.
2. The brim is not on the inside of the hole in the inner part.
3. The brim is on the outside of the inner part, and therefore
4. The brim is on the inside of the hole in which there is another part.

maksdampf, 2018-02-20:
Hi @BagelOrb , i don't think so.
1. Yes
2. Yes, but there are also some holes where there is a brim.
3. No, as i said this part is a solid. Both parts are connected by an arch a few layers upwards, thus there is no inner part.
4. No, it is a solid.

5. Even if the behaviour was any practical at all to control warping and bed adhesion through a partially enabled brim, i would not think that this is what the user expects from a feature called "Brim only on outside". I edited my previous post and described the expected behaviour in that.

Ghostkeeper, 2018-03-12:
> The catch is that it only removes the brim from holes that are empty.

Ghostkeeper, 2018-03-12:
After some reminiscing, remembering and deliberation, we've decided that the behaviour expected by the user is to remove all inside brims, not just the brims on the inside to holes which are empty. We could find no logical reason not to remove these brims if the user indicates that he wants to print only outside brims.

jackha, 2018-06-20:
Currently it's only "Brim Only On Outside", or everywhere. This is the current intended behavior and it's in the current version (3.4 beta).

closed 2018-06-20 by jackha