Ultimaker/Cura #1167: Cura not accepting .obj files

thefish548, 2016-11-22:
When I try to import a .obj file that I created on Autodesk Fusion 360, I get a failed to load error message. I really need to be able to edit and print this file, so please help as soon as possible. I use a Macbook air
[Swords of Sanghelios symbol (2).obj.zip](https://github.com/Ultimaker/Cura/files/606554/Swords.of.Sanghelios.symbol.2.obj.zip)

fieldOfView, 2016-11-22:
The problem is that the obj has normals but no uv coords. I vaguely remember, the OBJReader only supports objects with either no normals and no uv coordinates, only uv coordinates or both normals and uv coordinates. Only normals is not supported.

Because you're in a bind, I converted the object to an STL.
[Swords of Sanghelios symbol (2).zip](https://github.com/Ultimaker/Cura/files/606700/Swords.of.Sanghelios.symbol.2.zip)

Note that the object is also extremely small. Cura expects units to be mm, not meter. But Cura will autoscale the tiny object.

Ghostkeeper, 2017-04-24:
Our product owner just closed this because it has been inactive for long.

ianpaschal, 2018-05-22:
As part of our new ticket handling strategy, we voted as a team on the likelihood of fixing this issue. Unfortunately, we’ve decided that we will not take time to fix it, and are marking it as `Status: Won’t Fix/Do` and closing the issue.

closed 2018-05-22 by ianpaschal

fieldOfView, 2018-05-22:
Someone has a new copy-pasta...

ianpaschal, 2018-05-22:
Yup... voting as a team on whether to close or create JIRA ticket for any issues found the previous day is our new post-stand-up activity. Mixing in old GitHub issues too.

Ghostkeeper, 2018-06-28:
Just fixed this here: https://github.com/Ultimaker/Uranium/commit/dbb973e702701fd691574e42bc6067512ec38c66