Ultimaker/Cura #1168: Themes question

Arsakes, 2016-11-22:
What part of interface is styled using this themes :

`https://github.com/Ultimaker/Cura/tree/master/resources/themes/cura`
Is this used by main app window or by plugins  too?

Second questions:

Are all of the interface widgets declared using QtQuick QML or some parts of interface is created in other way?

fieldOfView, 2016-11-22:
The theme layer is used by all parts of Cura, including plugins. Note that almost everything in Cura is a plugin.

There are parts of Cura that use system styling. These are the parts that are not directly inside the main Cura window (ie: popups such as the preference panes).

AFAIK, all interface elements are declared using QML.

Arsakes, 2016-11-22:
Thanks, 

I also wonder how Cura knows which theme use (there is only one atm but still), there is no explicit mention of _/resources/themes/cura/styles.qml anywhere_ in the code or config file.

The other thing is QML does't have thing like **Theme.getFont()** or **Theme.getColor()** and such calls are mode from the styles.qml file. What's more its probably referencing the themes.json file which is next to styles.qml.

How it can be that i have access to **Theme** object withing _styles.qml?_
I'm checkcing QtQuick docs and QML syntax but I can't find such object.

fieldOfView, 2016-11-22:
Cura is based on the Uranium framework, and the Theme layer is part of Uranium.
https://github.com/Ultimaker/Uranium

The QML Theme object can be found here:
https://github.com/Ultimaker/Uranium/blob/master/UM/Qt/Bindings/Theme.py
It is registered to QML here:
https://github.com/Ultimaker/Uranium/blob/master/UM/Qt/Bindings/Bindings.py#L83

Cura itself also registers a number of objects to QML, eg here:
https://github.com/Ultimaker/Cura/blob/master/cura/CuraApplication.py#L550

Unless a preference is set for which theme to use, Uranium looks for a theme with the same name as the application identifier (which is "cura" for the Cura application):
https://github.com/Ultimaker/Uranium/blob/master/UM/Qt/Bindings/Theme.py#L40

Arsakes, 2016-11-22:
Was checking _http://ultimaker.github.io/Uranium/master/_ couldn't find it thanks!
If you can give lable to this issue plase label it as a questions

fieldOfView, 2016-11-22:
I have forfeited my labeling privileges...