Ultimaker/Cura #1169: In Cura 2.3.1 is CPE used. Why not PET of PETG

Leonav, 2016-11-22:
What is the difference between CPE, PEG/PETG?

Ghostkeeper, 2016-11-23:
They are different materials, as far as I understood.

CPE stands for Chlorinated PolyEthylene.
PEG stands for PolyEthylene Glycol.

Granted, they are both chains of methyl groups, but just about anything in organic chemistry is chains of methyl groups.

Montiey, 2016-11-26:
Hehe..
But yes, I think taking a day to google the basic specs for some oddball materials would help people out.

P.S. I'm new to GitHub. Is this the sort of thing I could help with somehow?

Ghostkeeper, 2016-11-28:
You can add PETG as a material via a pull request. It goes roughly like this:
* You create the material file on your computer (these are the `.fdm_material` files in the installation folder).
* You test the material file and find appropriate settings to use with PETG. There are 7 settings that can be altered in a material profile. They are here: https://github.com/Ultimaker/Cura/blob/master/plugins/XmlMaterialProfile/XmlMaterialProfile.py#L571 We'd like to include the option to modify any Cura setting later, but for now you'll have to do with these 7 (the .fdm_material format is meant to be an industry standard).
* You create a fork of https://github.com/Ultimaker/fdm_materials . This is where Ultimaker stores its materials that are shared among Cura and the firmware of their printers. You put the .fdm_material file there.
* You create a pull request to include your material in Ultimaker's repository. To create a pull request, you can reference [this guide](https://help.github.com/articles/creating-a-pull-request/) (or Google it yourself).

closed 2017-07-27 by nallath