Ultimaker/Cura #1170: Lowering Print Speed increases Initial Layer Travel Speed

JanBrunner, 2016-11-22:
When one halves the overall Print Speed, settings automatically double the Initial Layer Travel Speed. This can of course mess up the first layer. I suspect there is a slash or bracket in the wrong place or similar.

Ghostkeeper, 2016-11-23:
The formula is this: `Initial Layer Travel Speed = Initial Layer Speed * Travel Speed / Print Speed`

The aim of this was to keep the ratio between print speed and travel speed the same in the initial layer. Since you reduced the print speed but not the travel speed, the ratio between print speed and travel speed doubled. Hence the initial layer travel speed also doubled.

What would you say is a better formula to determine the initial layer travel speed? Or should the normal travel speed also scale with the print speed?

JanBrunner, 2016-11-23:
Thanks for your reply! Is there a reason to keep the ratio between print and travel speed the same? I thought that travel speed should usually be kept high (enough) to prevent stringing but on the first layer it can mess up the print.

When I change print speed, many speeds are affected but not the initial layer print speed. Wouldn't it be more reasonable to keep the ratio between print speed and initial layer print speed fixed by default? Maybe set both the initial layer print and travel speeds to half their respective "normal speeds". Then the print/travel ratio is kept the same, too, whether necessary or not.

Ghostkeeper, 2016-11-23:
The reasoning was that the `Initial Layer Travel Speed` would also be reduced/increased if you reduce/increase the normal `Travel Speed`.

The ratio must still depend on `Initial Layer Speed`. That is a hard requirement since that setting acts as the simplification-setting for when the user doesn't want to bother with adjusting every little detail. Perhaps more logical is to just replace the `Print Speed` in the formula with a fixed value (50 or so). We've done this for similar setting groups too.

JanBrunner, 2016-11-23:
What about something like this?
<Initial Layer Speed = Print Speed * 0.5>
and keeping
<Initial Layer Travel Speed = Initial Layer Speed * Travel Speed / Print Speed>

That way, the print/travel ratios are still the same and there is no surprise for users who don't check the initial layer settings when lowering the print speed.

Replacing <Print Speed> by a reasonable number in the formula as you suggest would also be fine for me.

closed 2016-11-24 by Ghostkeeper