Ultimaker/Cura #1172: Merge models inconsistent?

peteruithoven, 2016-11-23:
I loaded two simple blocks and tried to merge them. I noticed that they where merged inconsistently:
Sometimes it merged in a way that creates extra walls on the inside, and might print the infill with both materials: 
[image]
And sometimes it split up one of the two. This seemed preferable, it even moves around the part in the other material to continue with the second part.
[image]

There seemed to be no clear correlation with selection order.

BagelOrb, 2016-11-23:
What version of Cura are you running?

BagelOrb, 2016-11-23:
Is that one model file with two different objects? Or is it two models files which are merged using right click -> Merge models?

If it's one model file, you might want to look at the mesh fixes settings.
If it's two model files than this looks very much like a bug which I am currently working on.

peteruithoven, 2016-11-23:
Apologies for leaving little information. 
I'm running cura 2.3.1 from thopiekar's ppa: http://ppa.launchpad.net/thopiekar/cura
They where two different models, indeed merged with right click > Merge models.

alekseisasin, 2018-05-08:
Hi @peteruithoven , it is already long time ago when you created this issue. I would like to know. Did you try merge function in the latest Cura version?

BagelOrb, 2018-05-08:
@alekseisasin just test it yourself man.