Ultimaker/Cura #1173: CURA open file comes in different size than file

bphillips99, 2016-11-23:
I am working in Rhino / Imperial units. When I open a file it comes in very small. 
(I am assuming it is converting the build units to metric.)
I understand that CURA works in metric. I have converted some files to metric and I 
still get the same result. I understand there is a work around by scaling the object
by 2054%. But it would be great to have CURA open files at the right size regardless
of the units it was created in. Or be able to assign unit preferences in CURA.

I'm printing on the Ultimaker 3 using version 2.3.1 and 2.3.

I have a British colleague here in the office who works in Solidworks and has an
Ultimaker 2. She is using CURA 15.04.6. She usually works in metric but for this
project has been working in Imperial. She has had no issues with CURA re-scaling
her files and she has not done anything different for her exports.

fieldOfView, 2016-11-23:
> it would be great to have CURA open files at the right size regardless of the units it was created in

Unfortunately the STL format does not contain information about the unit the file was created in.

Cura expects the units to be mm. It could be that you are converting to metric meters instead of millimeters. In this case the objects come in at a factor of 1000 too small.

nallath, 2016-11-23:
3MF does have the ability to work with units. Cura should automatically convert that to metric (mm). 

For all other cases; I've made a plugin that fixes this (it scales models from inches to mm). It can be found on https://github.com/nallath/BarbarianPlugin/

closed 2016-11-23 by nallath