Ultimaker/Cura #1174: [Question] How to make sure start Gcode comes first

boelle, 2016-11-23:
Hi

trying out the 2.3.1 version and i have added some start G code for my printer

```
;FLAVOR:RepRap
;TIME:6280
;Generated with Cura_SteamEngine 2.3.1
M190 S60
M104 S200
M109 S200
M80;Power On
G28
G21 ; metric values
G90 ; absolute positioning
G92 E0 ; zero the extruded length
G1 X90 Y180 F6000 ; go to the middle of the front
G1 Z0.05 ; close to the bed
M104 S{print_temperature} ; set extruder temp
M190 S{print_bed_temperature} ; set bed temp & wait
M109 S{print_temperature} ; wait for extruder temp
G1 X140 E10 F200 ; make a thick line to prime extruder
G92 E0 ; zero the extruded length
G1 Z0.3 ; lift Z
;LAYER_COUNT:160
;LAYER:0
```

as you can see Cura adds something before that and i get a heating failed as the PSU is not turned on as the first thing... worked fine in 15.04.6

is there any simple way i can get the M80 First ?

fieldOfView, 2016-11-23:
In your start gcode, make sure to use ```{material_print_temperature}``` instead of ```{print_temperature}``` and ```{material_bed_temperature}``` instead of ```{print_bed_temperature}```. That should prevent CuraEngine from prepending the print and bed temperatures on its own.

boelle, 2016-11-23:
oki, will check... just to push it is there any way i can avoid the extra M104 S0 in the end code?

boelle, 2016-11-23:
that went away also... so it must be because it does not prepend the start code

closed 2016-11-23 by boelle

Ghostkeeper, 2016-11-23:
It assumes that the start g-code sets the print temperature itself if it requests `{material_print_temperature}` and doesn't prepend the nozzle heating g-code then. Similar for bed temperatures.