Ultimaker/Cura #1175: custom dual head printer slice...

hexxter, 2016-11-24:
If i use my custom dual head printer settings and load a stl file then the slice... runs for ever.
No error log in the console (no progressbar is shown). When i change to the one head version slicing finish after a short time.

BR Hexxter

fieldOfView, 2016-11-24:
Let me use your publicly available information to break into your computer network, find your Cura installation and get your custom dual printer settings so we can reproduce this. Won't be long...

...

No, I cannot actually do that. A bit more information would help. If you created or edited files, please share them so we can check them.

hexxter, 2016-11-24:
```
/extruders/
custom_dual_extruder_left.def.json

    "id": "custom_dual_extruder_left",
    "version": 2,
    "name": "1st Extruder",
    "inherits": "fdmextruder",
    "metadata": {
        "machine": "custom_dual",
        "position": "0"
    },

    "overrides": {
        "extruder_nr": {
            "default_value": "0",
            "maximum_value": "1"
        },
        "machine_nozzle_offset_x": { "default_value": 0.0 },
        "machine_nozzle_offset_y": { "default_value": 0.0 },

        "machine_extruder_start_pos_abs": { "default_value": true },
        "machine_extruder_start_pos_x": { "value": "prime_tower_position_x" },
        "machine_extruder_start_pos_y": { "value": "prime_tower_position_y" },
        "machine_extruder_end_pos_abs": { "default_value": true },
        "machine_extruder_end_pos_x": { "value": "prime_tower_position_x" },
        "machine_extruder_end_pos_y": { "value": "prime_tower_position_y" }
    }
}

```
```
custom_dual_extruder_right.def.json
{
    "id": "custom_dual_extruder_right",
    "version": 2,
    "name": "2nd Extruder",
    "inherits": "fdmextruder",
    "metadata": {
        "machine": "custom_dual",
        "position": "1"
    },

    "overrides": {
        "extruder_nr": {
            "default_value": "1",
            "maximum_value": "1"
        },
        "machine_nozzle_offset_x": { "default_value": 0.0 },
        "machine_nozzle_offset_y": { "default_value": 0.0 },

        "machine_extruder_start_pos_abs": { "default_value": true },
        "machine_extruder_start_pos_x": { "value": "prime_tower_position_x" },
        "machine_extruder_start_pos_y": { "value": "prime_tower_position_y" },
        "machine_extruder_end_pos_abs": { "default_value": true },
        "machine_extruder_end_pos_x": { "value": "prime_tower_position_x" },
        "machine_extruder_end_pos_y": { "value": "prime_tower_position_y" }
    }
}
```

definitions/
```
custom_dual.def.json

{
    "id": "custom_dual",
    "version": 2,
    "name": "Custom dual",
    "inherits": "fdmprinter",
    "metadata": {
        "visible": true,
        "author": "Ultimaker",
        "manufacturer": "Custom",
        "category": "Custom",
        "file_formats": "text/x-gcode",
        "has_materials": true,
        "machine_extruder_trains":
        {
            "0": "custom_dual_extruder_left",
            "1": "custom_dual_extruder_right"
        },
        "first_start_actions": ["MachineSettingsAction"]
    }
}
```

PPA: deb http://ppa.launchpad.net/thopiekar/cura/ubuntu xenial main
ubuntu 16.04

fieldOfView, 2016-11-24:
The custom_dual.def.json needs an overrides section that includes (at least) a value for "machine_extruder_count": 
```
    "overrides": {
        "machine_extruder_count": { "default_value": 2 }
    }
```

hexxter, 2016-11-24:
thank you this was the solution ;o)

closed 2016-11-24 by hexxter