Ultimaker/Cura #1180: CTRL+UP/DOWN does not update the layer number next to the layer slider

OlafWinkler, 2016-11-26:
CTRL+UP/DOWN change the layer that is shown, but they do not update the layer slider nor the layer number next to the layer slider.

BerndJM, 2016-11-26:
Which OS and which Cura-version we're talking about ?!?

OlafWinkler, 2016-11-26:
This issue is on my Win7 64-bit (all updates installed) and Cura 2.3.1.

OlafWinkler, 2016-11-26:
Hello,

 

this issue is on my Win7 64-bit (all updates installed) and Cura 2.3.1.

Thanks,

 

Olaf

Ghostkeeper, 2016-11-28:
Could not reproduce here on Ubuntu with the master branch.

fieldOfView, 2016-11-28:
Note that pressing CTRL is not necessary. Once you give focus to the layer number box, cursor up/down should work

OlafWinkler, 2016-11-28:
Agree. I just found that most of the time, the layer number and slider do indeed update correctly. It was difficult to reproduce this problem.
But I found this (the second point is about my original issue):
* Sometimes CTRL+up/down do not work at all, e. g. after LMB clicking somewhere to deselect all print objects and RMB-hold-perspective change or after activating one print object, then changing the .gcode file name, pressing "Enter" and clicking somewhere to deactivate the print object (in the lower right, above the duration, meters and grams). Activating a print object and clicking on the "Move" modifier (upper left, above scale and rotate) recover the function.

* Got it now: After entering a specific layer number into the box (or clicking the box and using cursor up/down), then activating a print object and clicking on "Move", the box and the slider do not update any more with CTRL+up/down. Even clicking into the layer number box and pressing "Enter" do not get you back to that layer. You must enter a new number. You can delete the number in the box and it will update again.
* Or: Type in a specific layer number, press "Enter", press "Esc". Using CTRL+up/down works, but layer slider and number do not update any more. Instead of a number you can even put in a SPACE, then the box appears to be empty.

Two days ago I got a little lost with the box and slider not updating, not realizing that the specific number that I typed in caused this.

General: Clicking into the number box and using cursor up/down recovers proper function and is a really convenient function.
This Cura version works fast and reliably and generates really good gcode. Thanks a lot for this great software.

Olaf

Ghostkeeper, 2016-11-29:
I could reproduce the first and last cases. Not the second one yet. But indeed this is confirmed to be a bug. Thanks!

OlafWinkler, 2016-11-29:
Thanks for checking.
The thing with the second point is to stick exactly to the procedure:

After entering a specific layer number into the box and press "Enter" (or clicking the box and using cursor up/down), then activating a print object (i. e. click on the print model once or twice until the move arrows appear) and clicking on "Move" (click on the "Move" icon in the upper left), the box and the slider do not update any more with CTRL+up/down.

ianpaschal, 2018-02-27:
These no longer reproduce in Cura 3.2.1.

closed 2018-02-27 by ianpaschal