Ultimaker/Cura #1182: Manage Printers / Add Printer ScrollView problem

cederom, 2016-11-26:
When adding New Printer in Manage Printers, the "Other" machine configuration  list is truncated as the ScrollView does not seem to work correctly and the window cannot be resized. I had to put my machine into "Custom" to select it, otherwise it was not visible.
[image]

Ghostkeeper, 2016-11-28:
Indeed a bug.

You should be able to resize the window, though. At least it is possible on Ubuntu.

cederom, 2016-11-28:
cannot resize on osx.. :-(

fieldOfView, 2017-06-12:
This has been fixed a good while back, issue can be closed

closed 2017-06-13 by awhiemstra