Ultimaker/Cura #1186: [master] Cura doesn't start

thopiekar, 2016-11-27:
```
thopiekar@home:~$ cura_clean 
removed directory '/home/thopiekar/.local/share/cura/materials'
'/home/thopiekar/.local/share/cura/user/BQ+Hephestos+2_current_settings.inst.cfg' wurde entfernt
removed directory '/home/thopiekar/.local/share/cura/user'
removed directory '/home/thopiekar/.local/share/cura/extruders'
'/home/thopiekar/.local/share/cura/cura.log' wurde entfernt
removed directory '/home/thopiekar/.local/share/cura/variants'
removed directory '/home/thopiekar/.local/share/cura/plugins'
removed directory '/home/thopiekar/.local/share/cura/quality'
'/home/thopiekar/.local/share/cura/machine_instances/BQ+Hephestos+2.stack.cfg' wurde entfernt
removed directory '/home/thopiekar/.local/share/cura/machine_instances'
removed directory '/home/thopiekar/.local/share/cura/'
'/home/thopiekar/.config/cura/cura.cfg' wurde entfernt
removed directory '/home/thopiekar/.config/cura/'
'/home/thopiekar/.cache/cura/definitions/2.3.99-master/zone3d_printer' wurde entfernt
'/home/thopiekar/.cache/cura/definitions/2.3.99-master/ultimaker2_extended' wurde entfernt
'/home/thopiekar/.cache/cura/definitions/2.3.99-master/ord_extruder_3' wurde entfernt
'/home/thopiekar/.cache/cura/definitions/2.3.99-master/printrbot_play' wurde entfernt
'/home/thopiekar/.cache/cura/definitions/2.3.99-master/bq_hephestos_2_marlin' wurde entfernt
'/home/thopiekar/.cache/cura/definitions/2.3.99-master/ultimaker3' wurde entfernt
'/home/thopiekar/.cache/cura/definitions/2.3.99-master/prusa_i3_mk2' wurde entfernt
'/home/thopiekar/.cache/cura/definitions/2.3.99-master/ultimaker_original' wurde entfernt
'/home/thopiekar/.cache/cura/definitions/2.3.99-master/punchtec_connect_xl' wurde entfernt
'/home/thopiekar/.cache/cura/definitions/2.3.99-master/bq_witbox' wurde entfernt
'/home/thopiekar/.cache/cura/definitions/2.3.99-master/ultimaker_original_dual_2nd' wurde entfernt
'/home/thopiekar/.cache/cura/definitions/2.3.99-master/rigidbot_big' wurde entfernt
'/home/thopiekar/.cache/cura/definitions/2.3.99-master/ultimaker_original_dual' wurde entfernt
'/home/thopiekar/.cache/cura/definitions/2.3.99-master/ultimaker2' wurde entfernt
'/home/thopiekar/.cache/cura/definitions/2.3.99-master/ultimaker_original_dual_1st' wurde entfernt
'/home/thopiekar/.cache/cura/definitions/2.3.99-master/printrbot_play_heated' wurde entfernt
'/home/thopiekar/.cache/cura/definitions/2.3.99-master/ultimaker3_extended' wurde entfernt
'/home/thopiekar/.cache/cura/definitions/2.3.99-master/ultimaker_original_plus' wurde entfernt
'/home/thopiekar/.cache/cura/definitions/2.3.99-master/rigid3d_zero' wurde entfernt
'/home/thopiekar/.cache/cura/definitions/2.3.99-master/rigidbot' wurde entfernt
'/home/thopiekar/.cache/cura/definitions/2.3.99-master/m180' wurde entfernt
'/home/thopiekar/.cache/cura/definitions/2.3.99-master/fdmextruder' wurde entfernt
'/home/thopiekar/.cache/cura/definitions/2.3.99-master/ultimaker3_extruder_right' wurde entfernt
'/home/thopiekar/.cache/cura/definitions/2.3.99-master/ultimaker2_extended_plus' wurde entfernt
'/home/thopiekar/.cache/cura/definitions/2.3.99-master/grr_neo' wurde entfernt
'/home/thopiekar/.cache/cura/definitions/2.3.99-master/ord_extruder_4' wurde entfernt
'/home/thopiekar/.cache/cura/definitions/2.3.99-master/ultimaker2_go' wurde entfernt
'/home/thopiekar/.cache/cura/definitions/2.3.99-master/robo_3d_r1' wurde entfernt
'/home/thopiekar/.cache/cura/definitions/2.3.99-master/punchtec_connect_xl_extruder_right' wurde entfernt
'/home/thopiekar/.cache/cura/definitions/2.3.99-master/ultimaker3_extruder_left' wurde entfernt
'/home/thopiekar/.cache/cura/definitions/2.3.99-master/bq_hephestos' wurde entfernt
'/home/thopiekar/.cache/cura/definitions/2.3.99-master/ultimaker2_plus' wurde entfernt
'/home/thopiekar/.cache/cura/definitions/2.3.99-master/bq_hephestos_2' wurde entfernt
'/home/thopiekar/.cache/cura/definitions/2.3.99-master/kupido' wurde entfernt
'/home/thopiekar/.cache/cura/definitions/2.3.99-master/uniqbot_one' wurde entfernt
'/home/thopiekar/.cache/cura/definitions/2.3.99-master/ultimaker3_extended_extruder_right' wurde entfernt
'/home/thopiekar/.cache/cura/definitions/2.3.99-master/julia' wurde entfernt
'/home/thopiekar/.cache/cura/definitions/2.3.99-master/custom' wurde entfernt
'/home/thopiekar/.cache/cura/definitions/2.3.99-master/bfb' wurde entfernt
'/home/thopiekar/.cache/cura/definitions/2.3.99-master/innovo_inventor' wurde entfernt
'/home/thopiekar/.cache/cura/definitions/2.3.99-master/ord' wurde entfernt
'/home/thopiekar/.cache/cura/definitions/2.3.99-master/punchtec_connect_xl_extruder_left' wurde entfernt
'/home/thopiekar/.cache/cura/definitions/2.3.99-master/deltabot' wurde entfernt
'/home/thopiekar/.cache/cura/definitions/2.3.99-master/prusa_i3' wurde entfernt
'/home/thopiekar/.cache/cura/definitions/2.3.99-master/mendel90' wurde entfernt
'/home/thopiekar/.cache/cura/definitions/2.3.99-master/rigid3d_3rdgen' wurde entfernt
'/home/thopiekar/.cache/cura/definitions/2.3.99-master/printrbot_simple' wurde entfernt
'/home/thopiekar/.cache/cura/definitions/2.3.99-master/ord_extruder_2' wurde entfernt
'/home/thopiekar/.cache/cura/definitions/2.3.99-master/rigid3d_hobby' wurde entfernt
'/home/thopiekar/.cache/cura/definitions/2.3.99-master/bq_witbox_2' wurde entfernt
'/home/thopiekar/.cache/cura/definitions/2.3.99-master/ultimaker' wurde entfernt
'/home/thopiekar/.cache/cura/definitions/2.3.99-master/maker_starter' wurde entfernt
'/home/thopiekar/.cache/cura/definitions/2.3.99-master/mankati_fullscale_xt_plus' wurde entfernt
'/home/thopiekar/.cache/cura/definitions/2.3.99-master/prusa_i3_xl' wurde entfernt
'/home/thopiekar/.cache/cura/definitions/2.3.99-master/kossel_mini' wurde entfernt
[... 31 lines cut ...]
2016-11-27 16:31:54,123 - CRITICAL - cura.CrashHandler.show [32]: Traceback (most recent call last):
2016-11-27 16:31:54,123 - CRITICAL - cura.CrashHandler.show [32]:   File "/usr/lib/python3/dist-packages/UM/PluginRegistry.py", line 252, in _populateMetaData
2016-11-27 16:31:54,123 - CRITICAL - cura.CrashHandler.show [32]:     meta_data = plugin.getMetaData()
2016-11-27 16:31:54,123 - CRITICAL - cura.CrashHandler.show [32]: AttributeError: module 'plugins' has no attribute 'getMetaData'
2016-11-27 16:31:54,124 - CRITICAL - cura.CrashHandler.show [32]: 
2016-11-27 16:31:54,124 - CRITICAL - cura.CrashHandler.show [32]: During handling of the above exception, another exception occurred:
2016-11-27 16:31:54,124 - CRITICAL - cura.CrashHandler.show [32]: Traceback (most recent call last):
2016-11-27 16:31:54,124 - CRITICAL - cura.CrashHandler.show [32]:   File "/usr/bin/cura", line 61, in <module>
2016-11-27 16:31:54,124 - CRITICAL - cura.CrashHandler.show [32]:     app = cura.CuraApplication.CuraApplication.getInstance()
2016-11-27 16:31:54,125 - CRITICAL - cura.CrashHandler.show [32]:   File "/usr/lib/python3/dist-packages/UM/Application.py", line 317, in getInstance
2016-11-27 16:31:54,125 - CRITICAL - cura.CrashHandler.show [32]:     Application._instance = cls()
2016-11-27 16:31:54,125 - CRITICAL - cura.CrashHandler.show [32]:   File "/usr/lib/python3/dist-packages/cura/CuraApplication.py", line 174, in __init__
2016-11-27 16:31:54,125 - CRITICAL - cura.CrashHandler.show [32]:     super().__init__(name = "cura", version = CuraVersion, buildtype = CuraBuildType)
2016-11-27 16:31:54,126 - CRITICAL - cura.CrashHandler.show [32]:   File "/usr/lib/python3/dist-packages/UM/Qt/QtApplication.py", line 88, in __init__
2016-11-27 16:31:54,126 - CRITICAL - cura.CrashHandler.show [32]:     self._loadPlugins()
2016-11-27 16:31:54,126 - CRITICAL - cura.CrashHandler.show [32]:   File "/usr/lib/python3/dist-packages/cura/CuraApplication.py", line 437, in _loadPlugins
2016-11-27 16:31:54,126 - CRITICAL - cura.CrashHandler.show [32]:     self._plugin_registry.loadPlugins()
2016-11-27 16:31:54,126 - CRITICAL - cura.CrashHandler.show [32]:   File "/usr/lib/python3/dist-packages/UM/PluginRegistry.py", line 123, in loadPlugins
2016-11-27 16:31:54,127 - CRITICAL - cura.CrashHandler.show [32]:     plugin_data = self.getMetaData(plugin_id)
2016-11-27 16:31:54,127 - CRITICAL - cura.CrashHandler.show [32]:   File "/usr/lib/python3/dist-packages/UM/PluginRegistry.py", line 143, in getMetaData
2016-11-27 16:31:54,127 - CRITICAL - cura.CrashHandler.show [32]:     if not self._populateMetaData(plugin_id):
2016-11-27 16:31:54,127 - CRITICAL - cura.CrashHandler.show [32]:   File "/usr/lib/python3/dist-packages/UM/PluginRegistry.py", line 255, in _populateMetaData
2016-11-27 16:31:54,127 - CRITICAL - cura.CrashHandler.show [32]:     raise InvalidMetaDataError(plugin_id)
2016-11-27 16:31:54,128 - CRITICAL - cura.CrashHandler.show [32]: UM.PluginError.InvalidMetaDataError: Invalid metadata for plugin plugins
```

fieldOfView, 2016-11-27:
Only way I can reproduce this is by creating a folder named "plugins" inside the plugins folder and adding an empty __init__.py file to it. Are you sure you don't have a nested plugins folder somewhere?

thopiekar, 2016-11-27:
Let me check

thopiekar, 2016-11-27:
Nope

thopiekar, 2016-11-27:
When working on the packaging this weekend I noticed one new file:
/usr/lib/cura/plugins/__init__.py

I also checked my packages: There is no plugins/plugins/__init__.py anywhere.

thopiekar, 2016-11-27:
Hmm, running Cura from sources via command-line works, but running the "normal" (PPA) installation does not.

awhiemstra, 2016-11-28:
This was an issue introduced by @nallath when moving the 3MF Writer plugin. Fixed with https://github.com/Ultimaker/Cura/commit/554a8a2e6a802d25816d4f0746f470f060bc0c56

closed 2016-11-28 by awhiemstra