Ultimaker/Cura #1187: malformed printer configuration causes cura to crash

cederom, 2016-11-27:
`2016-11-27 23:20:12,876 - CRITICAL - /Users/ultimaker/build/2.3/build/inst/lib/python3.5/site-packages/cura/CrashHandler.py (show [24]): RecursionError: maximum recursion depth exceeded in comparison
qmlRegisterSingletonType(): "ProfilesModel" is not available because the callback function returns a null pointer.
Abort trap: 6`

cederom, 2016-11-27:
Also malformed configuration can be somehow stored in the printer list. Then it is impossible to remove that printer from a list as the application crashes:

```
Version: 2.3.1
Platform: Darwin-16.1.0-x86_64-i386-64bit
Qt: 5.6.1
PyQt: 5.6

Exception:
Traceback (most recent call last):
  File "/Users/ultimaker/build/2.3/build/inst/lib/python3.5/site-packages/cura/Settings/MachineManager.py", line 1046, in getDefinitionByMachineId
AttributeError: 'NoneType' object has no attribute 'getId'
```

awhiemstra, 2016-11-28:
Yeah the problem here is that we set the active printer in the constructor of cura/Settings/MachineManager . If that constructor fails it fails to return a proper instance of that class triggering a loop. I have been meaning to fix that but never got around to it.

ChrisTerBeke, 2017-11-14:
Is this still an issue?

ianpaschal, 2018-03-26:
I can't reproduce it in 3.3 by mangling my printer configs, so I'm marking it as fixed.

closed 2018-03-26 by ianpaschal