Ultimaker/Cura #1190: After succesful print job cura not want print again ower network

landway, 2016-11-28:
Hello everyone,

after a successful print I Remove object and I confirm notification on display about removing print. Than I did another printjob in CURA 2.3.1. Now I want print again ower network and CURA says that printer is bussy. But printer is not busy. I tried restart printer. Nothing by network.

Ghostkeeper, 2016-11-28:
Cannot reproduce this with the steps given.

What does it say when you go into the browser to the following address:
    http://<printer's IP address>/api/v1/printer/status

You can find the IP address on the printer by going into maintenance, then network.

landway, 2016-11-28:
now it says "idle"

Ghostkeeper, 2016-11-28:
Strange. That's what the code checks for: Whether it is not "idle".

fieldOfView, 2016-11-28:
The only logical conclusion is that Cura stops checking. 
@landway: When this happens, does the monitor icon in the top-right corner still have a status overlay (green or blue)? Can you post your cura.log file to a pastebin?

This has been reported before, but the issue went away. See here: https://github.com/Ultimaker/Cura/issues/1110

landway, 2016-11-28:
@fieldOfView green .... I can not post logs because github say we don´t support that file type... (my file type is *.ZIP :D)

By the way, this problem is with printer UM3E where I have another problem with connecting with CURA. (if I have both printers ON - UM3 and UM3E. Than when i choose UM3E in CURA - UM3 displays notice with allow access...)

Ghostkeeper, 2016-11-28:
Yes we've seen that issue recently too and are in the process of fixing that. It seems to connect to the wrong printer.

landway, 2016-11-28:
[cura.txt](https://github.com/Ultimaker/Cura/files/616597/cura.txt)
[stderr.txt](https://github.com/Ultimaker/Cura/files/616598/stderr.txt)
[ultimakersystem-ccbdd30001ac.3.4.16.20161104.boot0.txt](https://github.com/Ultimaker/Cura/files/616599/ultimakersystem-ccbdd30001ac.3.4.16.20161104.boot0.txt)
[ultimakersystem-ccbdd30001ac.3.4.16.20161104.boot-1.txt](https://github.com/Ultimaker/Cura/files/616596/ultimakersystem-ccbdd30001ac.3.4.16.20161104.boot-1.txt)
[ultimakersystem-ccbdd30001ac.3.4.16.20161104.boot-2.txt](https://github.com/Ultimaker/Cura/files/616600/ultimakersystem-ccbdd30001ac.3.4.16.20161104.boot-2.txt)

Ghostkeeper, 2016-11-29:
@landway The issue where it connects with the wrong printer has been attributed to switching the wifi network of your computer. If that was the case for you, making sure that the computer doesn't accidentally switches wifi is sort-of a workaround for now.

partsoven, 2016-12-01:
My printer is hardwired to my network, not connected to Wifi. This bug is frustraing. Hope it gets squashed soon.
I will try to connect to Wifi and see if this helps...

landway, 2017-01-05:
Still faced with this problem. Anyone have any solutions or progress?

After succesfully wifi print, I remove print from platform, I certify on display removing print - than try again wifi print and CURA 2.4.0 say printer bussy...

This time it's a new situation, because this all I try with new CURA 2.4.0. and with new FW 3.5.3.20161221 - unlike past.

partsoven, 2017-01-05:
Yep can confirm still happening. I have 2 UM3's and this problem is very frustrating.

Floatdesignlab, 2017-01-07:
I was able to fix this by going into the -Printers- drop down and then Selecting -Manage Printers- then under the printer you would like to print to select -Activate-. 
You might have to do this each time until a real fix is made.

Ghostkeeper, 2017-01-22:
If the desired printer is not active in Cura (i.e. a different printer is selected), how are you trying to print something to the desired printer? Or is it actually displaying the desired printer as the active printer before you click "Activate"?

Floatdesignlab, 2017-01-22:
@Ghostkeeper in my case I only have one printer available. That printer is actually displaying as the active printer before I click activate. Prior to selecting activate, the printer showed as if it were active but had no connectivity. Once I select activate the printer becomes connected again.

Ghostkeeper, 2017-02-24:
This was fixed in 2.4.

closed 2017-02-24 by Ghostkeeper