Ultimaker/Cura #1199: Filament profiles go missing

themaninthesuitcase, 2016-11-29:
Application Version: 2.3.1
Platform: OS X 10.11.6

Steps to Reproduce:
Set up some custom filament profiles
Close Cura
Open Cura

Actual Results:
Filament profiles go missing, if one is currently selected it remains.  Not 100% of open/close cycles (i.e. they stay if I close and re open) but certainly happened 3 times now.

Expected results:
Custom filament profiles should persist.

Additional Information:
Cura.log attached as curalog.txt

themaninthesuitcase, 2016-11-29:
I just looked in ~/Application Support/Cura/materials and there's a lot of profiles there, so they exist they just aren't loading.

I've noticed they all have the same GUID in the meta data, is this problem? I created them by duplicating the Generic PLA then changing the names, colour, temp etc.

Slightly off topic, if I set the dia to say 1.73 should this be honoured but the main front end when I select it? Currently this seems to have no effect.

[image]

BerndJM, 2016-11-29:
Mmmmh ...
seems it's the same bug as this one: https://github.com/Ultimaker/Cura/issues/1104
Could you checkplease, if the temporary "fix" works also for you?

themaninthesuitcase, 2016-11-29:
👍 Yup that "fixes" it.
I did do a quick search, but apparently not a good enough one.

closed 2016-11-29 by themaninthesuitcase