Ultimaker/Cura #1200: SD-card / hotplug not detected on Windows 10

JasperAlgra, 2016-11-29:
Cura 2.3.1 (and previous 2.3.0 beta) are not detecting my SD card, thus no option to "save to removeable drive".  The card is detected by windows, it pop-up under explorer as a removable drive. Also unplugging the reader creates the 'usb connect' sound.

- Using cardreader in different usb ports
- different SD cards (original UM2+ card or other) 
- Unplugging/plugging while cura is running (card and/or reader). 
- Unplugging/plugging before cura is running (card and/or reader). 

doesn't seem to make a difference.

Windows 10 pro 64 bit.
Can I do anything to debug/help on this issue?

thopiekar, 2016-11-29:
Have you tried a different card-reader? I'm not familiar with the code for Windows, but it seems that your card-reader is identifying itself as an different class of device (just guessing the code for Windows is able to filter out eg. external hard-drives).

Ghostkeeper, 2016-11-30:
It looks in the Windows registry for the folder: `HKEY_LOCAL_MACHINE/HARDWARE/DEVICEMAP/SERIALCOMM`. Anything that has "USBSER" in the registry value is listed as a removable drive in Cura.

If you're willing to help, you could list for us the name and data of the keys in that folder?
You can find it by pressing the Windows key. Then type "regedit". Navigate to the folder I told about, and give us the names and values of all keys you see there.

I cannot reproduce this here, so it is either hardware dependent or it is because I'm on Windows 8.1 instead of 10.

fieldOfView, 2016-11-30:
@Ghostkeeper: those are serial ports for Usb printing, not removable drives

thopiekar, 2016-11-30:
@Ghostkeeper I've never used 8.1, but SD detection on my Windows 10 works here.

JasperAlgra, 2016-11-30:
Allright, thanks for the thinking along so far. Looks to be a local problem, I've just installed Cura 2.3.1 on my win10 work-pc and there it's working normally. 

I'll try a different cardreader at home tonight and post the results here. 

Would be nice if it's working again, the saving with one button is a nice feature :)

fieldOfView, 2016-11-30:
I think the cardreader is reporting a wrong drivetype. Could you open a dos prompt, and give the output of the following command (with the card in the cardreader)?

```WMIC logicaldisk get /value```

If my guess is correct, the following command will return "No Instance(s) Available.":
```WMIC logicaldisk where "DriveType=2" get /value | find ":"```

JasperAlgra, 2016-11-30:
Unfortunately on of my collegues just happen to have borrowed my other cardreader so cannot try that tonight.  :(

However, @fieldOfView, I've tried both commands:
```WMIC logicaldisk get /value```

The Ultimaker SD-card, skipped the (long) rest here. But if it's useful I can post that too.
```Access=0
Availability=
BlockSize=
Caption=I:
Compressed=FALSE
ConfigManagerErrorCode=
ConfigManagerUserConfig=
CreationClassName=Win32_LogicalDisk
Description=Verwisselbare schijf
DeviceID=I:
DriveType=2
ErrorCleared=
ErrorDescription=
ErrorMethodology=
FileSystem=FAT32
FreeSpace=3940777984
InstallDate=
LastErrorCode=
MaximumComponentLength=255
MediaType=
Name=I:
NumberOfBlocks=
PNPDeviceID=
PowerManagementCapabilities=
PowerManagementSupported=
ProviderName=
Purpose=
QuotasDisabled=
QuotasIncomplete=
QuotasRebuilding=
Size=3956801536
Status=
StatusInfo=
SupportsDiskQuotas=FALSE
SupportsFileBasedCompression=FALSE
SystemCreationClassName=Win32_ComputerSystem
SystemName=WORKSTATION
VolumeDirty=FALSE
VolumeName=ULTIMAKER
VolumeSerialNumber=31636230
```

``` WMIC logicaldisk where "DriveType=2" get /value | find ":"```
```C:\Users\jasper>WMIC logicaldisk where "DriveType=2" get /value | find ":"
Caption=F:
DeviceID=F:
Name=F:
Caption=I:
DeviceID=I:
Name=I:
Caption=J:
DeviceID=J:
Name=J:```

In cura I see there's a plugin responsible for SD-card/removeable drives, but I cannot find that one on my system for logs or so. You guys have any idea what I could try other than another cardreader?

Ghostkeeper, 2016-11-30:
A USB stick should work just as well. If it detects the USB stick but not the SD card reader, we'll know that something is up with the SD card reader.

If it doesn't detect the USB stick either though, I wouldn't know...

JasperAlgra, 2016-11-30:
@Ghostkeeper 
Cool, thanks for the info! 
Just tried that and USB stick does work, save button changes into "Save to Removeable Drive". So most probably the cardreader, weird but many many thanks. I'll double check tomorrow when I've got my other reader back :)

JasperAlgra, 2016-12-07:
Allright, sorry for my delayed reply. But it seems to be the cardreader, i've just tried another model and it's working in cura as supposed. Thanks for your input guys and sorry for the bother.

closed 2016-12-07 by JasperAlgra

thopiekar, 2016-12-07:
No problem. I'm glad to hear you've found the reason yourself :wink: