Ultimaker/Cura #1201: Cure crash

manuelpc2002, 2016-11-29:
Version: 2.3.0
Platform: Windows-7-6.1.7601-SP1
Qt: 5.5.1
PyQt: 5.5.1

Exception:
Traceback (most recent call last):
  File "cura_app.py", line 61, in <module>
  File "D:\2.3\build\inst\lib\python3.4\site-packages\UM\Application.py", line 308, in getInstance
  File "D:\2.3\build\inst\lib\python3.4\site-packages\cura\CuraApplication.py", line 173, in __init__
  File "D:\2.3\build\inst\lib\python3.4\site-packages\UM\Qt\QtApplication.py", line 94, in __init__
  File "D:\2.3\build\inst\lib\python3.4\site-packages\UM\VersionUpgradeManager.py", line 107, in upgrade
  File "D:\2.3\build\inst\lib\python3.4\site-packages\UM\VersionUpgradeManager.py", line 281, in _upgradeFile
MemoryError

fieldOfView, 2016-11-29:
32 bit?

You could try if clearing the files in C:\Users\[your username]\AppData\local\cura works. If you do that, Cura will not have to try upgrading these files, which may leave you with just enough RAM to pass this step.

manuelpc2002, 2016-11-29:
It's 32bit.

I achieved to load cure.
It was the cure.log it was very huge (more than 1Gb). I deleted it and
created a new one empty and it's works.

Thanks

fieldOfView, 2016-11-29:
NB: the application is called "Cura" with an "a".

manuelpc2002, 2016-11-29:
So Sorry, the dictionary in Spanish
Thanks

Saludos

Ghostkeeper, 2016-11-30:
If removing the log helped, that is a good hint on what causes this problem. It tries to find out whether that log file is something that it needs to upgrade to a newer version.
I've added a guard that will hopefully fix this once and for all.