Ultimaker/Cura #1202: Cura 2.3.0 hangs on "slicing..."

Sydbarr, 2016-12-01:


Ghostkeeper, 2016-12-01:
It doesn't happen for me on Windows 8.1 using Cura 2.3.0. I switched from Ultimaker 2 Extended to Ultimaker Original.

What operating system are you using, and what steps should we undertake to try and reproduce your problem? Just... load any model and it hangs on "Slicing..."?

Sydbarr, 2016-12-02:
I'm using Cura on Vista, 32 bit. I've already try to switch to other machine, but nothing change.
I've just load the model (simply parallelepiped) and it hangs on "Slicing..."

fieldOfView, 2016-12-02:
Please go to Help -> Show Configuration Folder

In the explorer window that opens, there should be a stderr.log file. Please paste the contents of that file into www.pastebin.com or similar (gist.github.com, chopapp.com), and give us the resulting link. Hopefully this will show us what is happening on your computer.

Sydbarr, 2016-12-05:
here's the link: http://pastebin.com/StDy3bJY

awhiemstra, 2016-12-05:
Do you happen to have some kind of aggressive firewall software running? From the log, CuraEngine is unable to connect to the local socket we create, which would indicate some other piece of software interfering.

Sydbarr, 2016-12-05:
Yes, with firewall disabled the slicing is complete.
Many thanks

fieldOfView, 2016-12-05:
For our record, what firewall were you using?

Sydbarr, 2016-12-06:
Comodo firewall, free edition.

Sydbarr, 2016-12-07:
hi,

I'am using Comodo firewall, free edition.

By

&lt;reply@reply.github.com&gt;

 To: "Ultimaker/Cura" &lt;Cura@noreply.github.com&gt;

 Cc: "Author" &lt;author@noreply.github.com&gt;

 Data: 05/12/16 15:06

 

 

&mdash;

 

 
 
 ZE-Light e ZE-Pro: servizi zimbra per caselle con dominio email.it, per tutti i dettagli 
Clicca qui http://posta.email.it/caselle-di-posta-z-email-it/?utm_campaign=email_Zimbra_102014=main_footer/f
 
 Sponsor:
 Soluzioni di email hosting per tutte le esigenze: dalle caselle gratuite a quelle professionali su piattaforma Zimbra, da quelle su proprio dominio a quelle certificate PEC. Confronta le soluzioni
 Clicca qui: http://adv.email.it/cgi-bin/foclick.cgi?mid=13326&d=7-12

closed 2017-07-27 by nallath