Ultimaker/Cura #1203: Cura will not slice any files.

coustillac, 2016-12-02:
My company just purchased an Ultimaker 3 Extended and have set it up and printed sample prints. We were able to connect the printer via Ethernet but we are having issues with Cura. We are using Cura 2.3.1 64 bit. The printer is connected and I am even able to view the onboard camera, but when I load a model, it will not slice. The 3D object gets dropped on the build plate but in the bottom right it just says "slicing..." and the bar does not move. This is consistent for every model I use. I can manipulate the models on the build plate but the slicing never happens. I am unable to print anything. This issue needs to be fixed.

fieldOfView, 2016-12-02:
Go to Help -> Show Configuration Folder
In the explorer window that opens, there should be a stderr.log file. Please paste the contents of that file into www.pastebin.com, and give us the resulting link. Hopefully this will show us what is happening on your computer.

coustillac, 2016-12-02:
I cannot use pastebin on my work computer. Any alternative?

fieldOfView, 2016-12-02:
You could try se https://gist.github.com/

coustillac, 2016-12-02:
I used chop instead. Here is the url.  http://chopapp.com/#3fv1csw0

thopiekar, 2016-12-02:
Again this strange problem. I remember there was a user already in the past, who had this problem, too. Cura.exe says that CuraEngine.exe is missing, but CuraEngine.exe was there. I personally only see two options:
1. The error message shows a wrong path (unlikely)
2. The security settings are set very high and (signed) applications are just allowed to run applications, which are signed, too.

However, saw neither 1. nor 2. by my own eyes.

fieldOfView, 2016-12-02:
@coustillac: Could you check if C:\Program Files\Cura 2.3\ has a file CuraEngine.exe?

@thopiekar: 2 sounds likely.

coustillac, 2016-12-02:
I do not see CuraEngine.exe in my folder. This is clearly the problem but how do I fix it?

thopiekar, 2016-12-02:
Could you check your anti-virus software whether it detected CuraEngine.exe as an potential virus/malware?

coustillac, 2016-12-02:
We uninstalled then reinstalled and it seems all the files are where they should be. Still, Cura will not slice.

coustillac, 2016-12-02:
I am having our IT employees look into this. How exactly would they detect this?

thopiekar, 2016-12-02:
Good question. I would check the security suite first. There also should be ways to check the system logs.
I'm not a Windows expert, so I hope they have better ideas how to solve this.

Thank you for your help @coustillac !

coustillac, 2016-12-02:
Hey all. We ended up uninstalling Cura version 2.3.1 and installed 2.3 instead. This has fixed the issue and everything is running properly. Thanks for the help

fieldOfView, 2016-12-03:
That is helpful information, but a bit of a mystery. AFAIK very little has changed between 2.3 and 2.3.1

thopiekar, 2016-12-03:
@coustillac Could you please tell us about your security software? Are you using Kaspersky or McAfee, for example?

Ghostkeeper, 2016-12-03:
Didn't we implement the signing of the Windows build in between 2.3.0 and 2.3.1? Or was that later?

thopiekar, 2016-12-03:
Yes, it was around it, but we just signed the Installer afaik. Since we are in the Windows Store we sign the Cura.exe now, but if I'm not wrong we never signed the engine.
Hope my recent work on signing will be used for the next release, so installer, Cura.exe and CuraEngine.exe will be signed :wink:

Loleyouss, 2016-12-08:
Hello, I have the same problem with a Mac System and the 2.3 will not fix the issue. 
could you please help me.

fieldOfView, 2016-12-08:
logs. we need to see logs

nallath, 2016-12-08:
[image]
Not these kind though

Loleyouss, 2016-12-08:
Perhaps this one ;-)

<script src="https://gist.github.com/Loleyouss/3fb3f32b2b62a38321beb6dc2db12841.js"></script>

Loleyouss, 2016-12-08:
sorry, this one 
https://gist.github.com/Loleyouss/3fb3f32b2b62a38321beb6dc2db12841

nallath, 2016-12-09:
You have a firewall that blocks the local connection.

Loleyouss, 2016-12-09:
Thank you, nevertheless, i disable Intego before install Cura and le Mac OSX firewall is Off.
I will look for another cause

closed 2017-07-27 by nallath