Ultimaker/Cura #1205: how does Cura  start Cura-engine in the newest version of Cura soure code?

elviswoo, 2016-12-03:
I know the communication way between Cura and Cura-engine has been changed after the version of 15.06. But, i cannot find how does Cura call the Cura-engine and in which .py files Cura send parameters to Cura-engine , Is there anybody help me?  thanks!

fieldOfView, 2016-12-03:
Communication between Cura and CuraEngine is handled here:
https://github.com/Ultimaker/Cura/blob/master/plugins/CuraEngineBackend/CuraEngineBackend.py

A single CuraEngine process is launched that stays active in the background untill Cura is closed. The two applications communicate with eachother over a layer called "Arcus", see https://github.com/ultimaker/libArcus. Data is encoded using Protobuf.

nallath, 2016-12-03:
If you want so see what is actually sent, have a look at the "startSliceJob"

BagelOrb, 2017-07-23:
[quation answered]

closed 2017-07-23 by BagelOrb