Ultimaker/Cura #1206: Too small Birm

J0h4nP, 2016-12-03:
I do not know if this is a handling error or a bug?
I flipped it and the problem was solved for me.
[image]
[image]
[image]

Ghostkeeper, 2016-12-03:
We have no error handling that does that sort of thing. It looks like a bug to me.

Ghostkeeper, 2016-12-03:
Is the component you're slicing available somewhere?
This bug doesn't reproduce on other models I've tried.

J0h4nP, 2016-12-03:
https://www.dropbox.com/s/otwbo53buwjk8b5/Landskap.stl?dl=0

thopiekar, 2016-12-03:
I tried to reproduce here with my BH2 and get something different on master.
Instead it doesn't center the model and says that the model doesn't fit on the plate.
Centering it manually work and the model becomes yellow (so it fits) and slices.

thopiekar, 2016-12-03:
Apart from that, everything works here. I also had a similar effect like @J0h4nP , but that's why the model is not completely flat at the bottom. Enabling support creates the brim correctly of course.

PS: About the centering problem: I'll add a JIRA ticket plus video I made here.

Ghostkeeper, 2016-12-05:
The model is indeed a bit crooked. It only creates a brim around the part that is touching the bed.

Lay Flat seems to work to put the entire bottom against the bed.

closed 2016-12-05 by Ghostkeeper

thopiekar, 2016-12-05:
Added the centering issue to JIRA: CURA-3091

J0h4nP, 2016-12-05:
In summary: the error was somewhere between the keyboard and backrest :)

thopiekar, 2016-12-05:
Correct :sweat_smile: