Ultimaker/Cura #1209: Setting "Printing Temperature" to 0 should not insert an M109 S0 command

michaelnew, 2016-12-03:
From the help bubble, I get the impression that setting the print temp to 0 in Cura should let either the machine itself or the printer host manage the print temperature.

[image]

In my case I use an OctoPrint plugin that inserts the print temp into the startup gcode. Unfortunately Cura inserts a `M109 S0` command that runs after the startup script, which shuts off the heater and fails the print.

When I set a non-zero print temperature, Cura inserts both a `M109` and a `M104`, and when I set the print temp to zero it only inserts a `M109` command. The fact that the `M104` disappears makes me think that inserting the `M109` is a bug.

fieldOfView, 2016-12-04:
Nice catch. This is the culprit:
https://github.com/Ultimaker/CuraEngine/blob/master/src/gcodeExport.cpp#L840

writeTemperatureCommand is being called twice; once without, and once with the wait argument set to true. The check if the temperature equals the "previous" temperature is effectively skipped if wait == true. It should probably actually check if the temperature being set is actually 0, in which case writeTemperatureCommand should just return. Calling @BagelOrb!

As an aside, I'm intrigues by the OctoPrint plugin; got a link for that?

BagelOrb, 2016-12-06:
Actually there are settings especially for this, but I'm not sure whether these settings are accessible from the frontend:
Wait for nozzle heatup
and
Include material temperatures

You could include new defaults for these settings in your machine.def.json file by adding
```
"material_print_temp_wait": { "default": false },
"material_print_temp_prepend": { "default": false },
```

michaelnew, 2016-12-06:
@BagelOrb Thanks, that will work for me since I don't want Cura ever adding print temperatures, but I do think it should be something that happens when you set the print temp to 0, not necessarily a setting you'd toggle on and off.

If you set a print temperature in Cura you likely want the printer to wait until that temp is reached, so  M109 should be there, but if you set the temp to 0 (you want to manage it on the machine or in some other way, like in my case) you would not want to have an M109 S0 shutting off your hotend. The current behavior definitely doesn't match the help-bubble, and I think it should.

@fieldOfView [The plugin](https://github.com/michaelnew/Octoprint-Material-Settings) is very much experimental, but it works well enough that I use it exclusively. Feel free to ping me if you try it out and have any questions or anything.

Ghostkeeper, 2016-12-07:
Apart from the workaround by modifying the settings, should we fix the functionality that setting the printing temperature to 0 skips the temperature g-code also?

BagelOrb, 2016-12-07:
Yeah I think I should edit the description.

I can understand why you would want the print temperature commands at the
beginning to be removed, but why would you want all other print temperature
commands removed?

Cura has quite some temperature tweaking features, which would all be
negated in such a case.

<https://github.com/Ultimaker/Cura/issues/1209#issuecomment-265246739>, or mute
the thread

michaelnew, 2016-12-07:
Sorry, I wasn't very clear in my last comment. I just meant that in my own particular case I don't want Cura setting print temps. You're right; setting the print temp to 0 should not remove every temp command from the gcode (other than the starting temps).

My complaint is simply that setting the print temperature to 0 will put at `M109 S0` command at the start of the print, and it clearly shouldn't. Setting it to 0 means that I want to manage temperatures outside of Cura, and that `M109` prevents that from working. Outside of that I think the way it behaves is correct.

skorokithakis, 2016-12-13:
I am seeing the same issue. The description says "set to 0 to preheat manually" but Cura effectively inserts a "cooldown" command. It shouldn't.

vitormhenrique, 2017-01-06:
Where can I find my machine.def.json file? I had to create a custom machine definition because my printer wasn't listed. 
Is there a way to create a post processor to remove the gcode?

Ghostkeeper, 2017-01-22:
You can find your machine's .def.json file in your Cura's installation folder (usually `C:\Program Files\Cura` on Windows). In that folder, go into `resources/definitions` and find the file with the name of your printer.

You can remove the g-code with a post-processor, yes. Post-processing scripts go into the `plugins/PostProcessingPlugin/scripts` folder. You'd have to write the script yourself though. But allow me to give you something to start with (untested):

    from ..Script import Script
    class ManualTemperatureFix(Script):
        def getSettingDataString(self):
            return """{
                "name": "Manual Temperature Fix",
                "key": "ManualTemperatureFix",
                "metadata": {},
                "version": 2,
                "settings": {}
            }"""

        def execute(self, data):
            for index, layer in enumerate(data):
                lines = layer.split("\n")
                lines = [line for line in layer.split("\n") if not line.startswith("M109")]
                data[index] = "\n".join(lines)
            return data

Save this to a file called `C:\Program Files\Cura\plugins\PostProcessingPlugin\scripts\ManualTemperatureFix.py` and restart Cura. You should then be able to select this script as post-processing filter. It simply removes all occurrences of M109 from the g-code.

BagelOrb, 2017-01-23:
I'm thinking of adding a setting Enable Temperature Control.

That should also help with people who modded their UM2 to a dual extrusion machine.

holywen, 2017-04-11:
In case you might need to cool down the extruder and the bed, the user may want to add  "End Gcode" to do that, in this case, we should not comment those lines:

`from ..Script import Script
class ManualTemperatureFix(Script):
    def getSettingDataString(self):
        return """{
            "name": "Manual Temperature Fix",
            "key": "ManualTemperatureFix",
            "metadata": {},
            "version": 2,
            "settings": {}
        }"""

    def execute(self, data):
        for index, layer in enumerate(data):
            output = ""
            need_to_fix_temp = True
            for line in layer.split("\n"):
                if line.startswith(";End GCode"):
                    need_to_fix_temp = False

                if need_to_fix_temp and (line.startswith("M109") or line.startswith("M140")):
                    out_line = ";" + line
                    output = output + "\n" + out_line
                else:
                    output = output + "\n" + line
            data[index] = output
        return data`

The End GCode need to start with:
`;End GCode`

identify3d-joe, 2017-04-20:
I can confirm that this bug is still in 2.5. Since we are using a custom printer profile, this means that we must use the gcode post-process solution which is a pain since we have to enable it every time we load Cura. Is there an ETA for this to get fixed?

BagelOrb, 2017-04-21:
In Cura 2.5 there is an option Enable Temperature Control. If you turn that
off (for a specific extruder) the nozzle temperature of that extruder train
will not be set.

Cyberman-tM, 2017-05-05:
> In Cura 2.5 there is an option Enable Temperature Control. If you turn that off (for a specific extruder) the nozzle temperature of that extruder train will not be set.
Where, or in which version of Cura? (Assuming there are multiple versions of 2.5)

I can't find that option anywhere. There is a similar option mentioned, "Enable Nozzle Temperature Control", but aside from being mentioned by the "Auto Temperature" option, I can't find that either.

These options somehow didn't make it to the public release, I guess?

BagelOrb, 2017-05-06:
It's in the release, but it's a machine setting, so users cannot edit it.
You have to edit it in the .def.json file of your printer.

BagelOrb, 2017-07-23:
fixed

closed 2017-07-23 by BagelOrb