Ultimaker/Cura #1211: Unable to exit Move Model function

Barry37, 2016-12-04:
Cura (2.3.1) sticks on Move model (a multiple of 6 models). Moving the mouse just keeps moving the model. The X position this time is now 28766mm, so the models are way out of the print area (and invisible). The only way I've found to get out of this is to Force Quit (Mac OS X). It's not possible to type a value in the "Move" panel, or choose "Scale" etc.
I can't think of a reason for actually being able to move a model off the print area (and it then being auto sliced while unprintable).

nallath, 2016-12-05:
This is a bug that should be fixed for 2.4

closed 2016-12-05 by nallath

vdesmedt, 2017-03-04:
I have exactly the same issue in 2.4 but only with big objects.  I have attached one of the object with which I encounter the trouble.
[Body2.stl.zip](https://github.com/Ultimaker/Cura/files/818476/Body2.stl.zip)

Ghostkeeper, 2017-03-06:
Doesn't reproduce for me with that model. Perhaps it depends on the build volume if it occurs only with big models. What printer are you using?

vdesmedt, 2017-03-06:
UMO. I have checked the dimensions in my settings which are correctly set to roughly 200/200/200 mm
I'm running the OS X version of cura

Ghostkeeper, 2017-03-07:
Still doesn't reproduce :( Though I am running the most recent development environment, and on Ubuntu rather than OSX. Shall we hope that it's fixed in there?

One more thing we could try is to share a .curaproject file rather than an STL model. I doubt that'll make any difference though.