Ultimaker/Cura #1212: Menus and pulldowns disappear when trying to select them

dWoolridge, 2016-12-04:
Application Version: Cura 2.3.1
Platform: Windows 7 Professional, 64-bit, Service Pack 1
Qt: << The version of Qt used >>
PyQt: << The version of PyQt used >>
Display Driver: AMD Radeon HD 6900 series

When I single-click on the main menu (File / Edit/ Settings / etc.) or the cura pulldowns (Printer / Material / Profile) they will appear for a fraction of a second and then disappear.  Further investigation revealed that the menu/pulldown appears for as long as I hold down the mouse button and then disappears when I release the mouse button.  It also seems to be associated with mouse movement while clicking.  I cannot re-produce the problem if I intentionally hold the mouse perfectly still while clicking.  I've never noticed this with other Windows applications, and my "mousing" technique is not different with Cura.

Steps to Reproduce:
Single click on a menu item (File / Edit / View / Settings, etc) or a pulldown (Printer / Material / Profile) with the mouse pointer still moving.

Actual Results:
The menu item or pulldown will disappear when the mouse button is released

Expected results:
The menu item should stay visible for use.

Ghostkeeper, 2017-02-18:
The product manager for Cura at Ultimaker rejected this issue for being too unimportant. This means that the main developers can't spend time on it.

nallath, 2017-02-20:
It's also an issue that I don't think we can fix at all, as it's something that's deep inside QT (The main GUI framework that we use).

closed 2018-02-27 by ianpaschal