Ultimaker/Cura #1213: Cura 2.3.1 cannot import profile

mikosik, 2016-12-04:
Cura 2.3.1 cannot import profile that was exported in cura 2.1.
It fails with:

[image]

reproduced with cura 2.3.1 on Linux Mint 18

Ghostkeeper, 2016-12-05:
Ah, I thought this was specifically fixed in 2.3.1 (vs. 2.3.0)? @nallath do you know more about this?

nallath, 2016-12-05:
Uh. I think it was yes. Not a 100% sure if it was for 2.3.1 or 2.4 though.

gvJaime, 2016-12-16:
I have this issue with cura 2.3.1-2 on Fedora 25, with materials.

When I import some material file, it says it has been successfully imported, but it doesn't show up.

Ghostkeeper, 2016-12-17:
That's a different problem though. I know that one well.

The material file specifies which printers are supported and which are not supported. If the currently active printer is not supported according to the material, then the material does not show up (but it will show up in the menu when you have a supported type of printer selected).

It's a bit of a user interaction problem because as a user I'd typically expect to see a result when I import something.

gvJaime, 2016-12-17:
Well, now I know this is not the issue to comment on this. Can you refer me to an open issue, or should I open a new one for this particular subject?

My problem now it's turned into this: The UI doesn't provide any way to manage material compatibility. 

I've inspected the xml, and added a similar <machine> instance for the bq hephestos XL, but it still doesn't show up. It doesn't matter what I put in the "product=" field, the name or the id. It still doesn't recognize it. When making a custom printer, I also cannot use it's name to add it into the compatibility list.

Also, haven't found the folder where imported profiles are stored, so I cannot dissect the problem anymore.

Ghostkeeper, 2016-12-27:
I don't think there is an open issue regarding that yet. We have an issue about it in our internal tracker though.

Ghostkeeper, 2016-12-27:
For now, you could try setting `hardware compatible` to `yes` in the XML file.

mikosik, 2017-01-11:
I installed cura 2.4 (beta) from https://ultimaker.com/en/products/cura-software and profiles export/import is still buggy.
I exported profile using cura 2.4 and I got following file:
[testbug.curaprofile.zip](https://github.com/Ultimaker/Cura/files/698610/testbug.curaprofile.zip)

This file is not a text file as I would expect but zip file. It can be imported back by cura 2.4.
However my old profile files exported from cura 2.3 cannot be imported in cura 2.4.

What I suggest is to make export/import work on pure text files only.

My OS: Linux Mint 18 (Sarah)

Ghostkeeper, 2017-01-22:
> What I suggest is to make export/import work on pure text files only.

We had to make it a zip file because there are multiple active profiles for multi-extrusion printers.

mikosik, 2017-01-23:
Thanks for explanations.

However this makes my life miserable. I have around 50 curaprofiles files in text format and I'm used to modify them (as text files in normal editors), compare them using diff tools, keeping them in repository and checking with 'git diff' what has changed recently in my profile so it produced different print recently.

Also changing format of *.curaprofile file in incompatible way so file produced by x.y version of software is not readable by x.(y+1) is really misleading to end user. I upgraded cura software from 4.2 to 4.3 (minor version changed) and all of sudden all my curaprofiles files are incompatible with cura software.

What I would suggest is to:
 - introduce *.curaprofileset extension that is zip file with multiple *.curaprofile files
 - keep *.curaprofile extension as pure text files format
 - allow cura to export and import both curaprofile and curaprofileset

or...
If you don't want to mess again with extension you can as well leave *.curaprofile as zip file and introduce *.curaprofiletxt format that contains pure txt file.

or...
you can automatically export profiles as zip files only for mutl-extrusion profiles and keep producing pure text files for others using the same extension. Import can detect easily whether given file is zip file or text file and can adapt appropriately.

sedwards2009, 2017-01-23:
@mikosik Cura 2.4 should be able to import the older pure text format which was used in 2.3.

mikosik, 2017-01-23:
@sedwards2009 

Thanks. When will be 2.4 version available for download?

Ghostkeeper, 2017-01-24:
It's planned for the end of this month.

mikosik, 2017-03-09:
I've just installed cura 2.4.0 downloaded from https://ultimaker.com/en/products/cura-software and it still cannot import pure text format of curaprofile.

[image]

Ghostkeeper, 2017-03-09:
Hmm, well we marked it as fixed in our internal tracker but perhaps you found a case where it still fails then. I'll make a new issue for it.

mikosik, 2017-05-22:
so 2 months passed, cura 2.5 has been released in the meantime and the bug still exists.

If you keep all your devs conversations on github instead of internal (hidden) tracker then
 - I could follow discussion and be informed about progress of fixing (even if it was pushed to backlog I could at least know how to plan my future work) instead of pinging here
 - I could take part in discussion and suggest better solution if I noticed that my report was misunderstood
 - you could save your time you spent copying info between github and internal tracker
 - I could download night build as soon as it is available and test it for you for free

It looks like all those possibilities are wasted :(

nallath, 2017-05-22:
I agree. But the choice for this is not up to us unfortunately.

Ghostkeeper, 2017-05-24:
Ultimaker tracks all of its R&D issues on Jira, including material development, software, testing, hardware development, product management, website development, basically everything. This includes new and yet unannounced projects, so Jira can't be made public. And they want to keep these issues interconnected, so it can't be moved to Github for just software.

You're not missing out on a whole lot of discussions. Basically what happens for these issues is that they get put on a pile of issues that get partially sorted by importance. They then sit there for 3 months and then our product manager throws them away for being too old. After all, nothing happened to the issue so it must not be important, right? Right... even though the issue might have been very active on Github. If nobody stepped in at that point the issue gets thrown away for real. If they were important enough to get picked out they'll get put in a sprint and we can fix them.

The nightly build is something we still hope to make available to the public though. There's no harm in that.

Ghostkeeper, 2017-06-05:
Our product manager just removed this from our planning for being inactive for 3 months.

closed 2018-02-21 by ianpaschal