Ultimaker/Cura #1214: Cura not remembering my printer

kl4m4, 2016-12-04:
Hi,
Cura 2.3.1 on mac os 10.12.1 keeps asking me for my printer. Every time I have to select one of the printers (Prusa i3 Mk2) tweek buildplate dimensions (mine are few mm smaller), add some auxilary startup g-code. Then I use it normally, slice, print, change settings in my profile etc etc... But next time I start cura it asks me for printer again. Profile setting are preserved though.
Is there any way to see some logs? Some debug console? Whatever?

Cheers!
kl4m4

Ghostkeeper, 2016-12-05:
There's a log file. Here's how to find it: https://github.com/Ultimaker/Cura#logging-issues
You can also run Cura from a console, which makes the log immediately visible.

BerndJM, 2016-12-05:
@Ghostkeeper: could this be the same on MacOS as this #1098 on Ubuntu ?!?

fieldOfView, 2016-12-05:
We won't know untill we see logs

kl4m4, 2016-12-05:
Ok, checking out logs did the trick....
```
2016-12-04 21:42:36,322 - ERROR - /Users/ultimaker/build/2.3/build/inst/lib/python3.5/site-packages/UM/Logger.py (logException [53]): Exception: Could not deserialize container Prusa i3 Mk2
2016-12-04 21:42:36,323 - ERROR - /Users/ultimaker/build/2.3/build/inst/lib/python3.5/site-packages/UM/Logger.py (logException [53]): Traceback (most recent call last):
2016-12-04 21:42:36,323 - ERROR - /Users/ultimaker/build/2.3/build/inst/lib/python3.5/site-packages/UM/Logger.py (logException [53]):   File "/Users/ultimaker/build/2.3/build/inst/lib/python3.5/site-packages/UM/Settings/ContainerRegistry.py", line 245, in load
2016-12-04 21:42:36,323 - ERROR - /Users/ultimaker/build/2.3/build/inst/lib/python3.5/site-packages/UM/Logger.py (logException [53]):   File "/Users/ultimaker/build/2.3/build/inst/lib/python3.5/site-packages/UM/Settings/ContainerStack.py", line 262, in deserialize
2016-12-04 21:42:36,323 - ERROR - /Users/ultimaker/build/2.3/build/inst/lib/python3.5/site-packages/UM/Logger.py (logException [53]): Exception: When trying to deserialize Prusa i3 Mk2, we received an unknown ID (prusa_i3_moj 0) for container
```
I had my printing profile named `moj 0,5mm` - it contained a comma. So in `/machine_instances/___.stack.cfg` it produced a line:
`containers = Prusa i3_current_settings,prusa_i3_moj 0,5mm,low,generic_pla,Prusa i3_variant,prusa_i3,` which cannot be deserialized due to that one extra container. Not sure how exactly it works (what info in which file) but when I renamed my printing profile to something without commas, everything works fine.
Thanks for pointing in right direction. For non-IT people it is not so obvious where to look for what.
Cheers!

fieldOfView, 2016-12-06:
A nice spot of debugging. We should probably disallow commas in some sensitive text entry fields.

Ghostkeeper, 2016-12-06:
Or escape them when creating an ID from it.

Ghostkeeper, 2016-12-06:
I could reproduce this (simply create a profile with a comma in the name and restart). I created an issue in our internal tracker to get this fixed, because it is quite critical.

closed 2016-12-08 by jackha