Ultimaker/Cura #1217: Wrong order dual printing 1st layer, tower

karabas, 2016-12-05:
Problem on the 1st layer dual printing/ tower, 1 skirt loop.
1  1st loop of skirt printed
2 detail printed
3 extruders switched
4  2nd loop of skirt printed
5 detail printed!!! insteed of tower
6 tower printed ( the same color)

Upper layers are fine.

BagelOrb, 2016-12-06:
On the first layer the skirt was counted as being good enough of a priming, so that it doesn't need to prime tower to be printed before the model. This has however been changed. In the next release the prime tower will be printed before the model on the first layer.

Do note that for dual color models which have both colors on the first layer, the prime tower cannot be printed before each color, because we can only print the prime tower once each layer.

I think the order is now
- skirt
- prime tower
- model
- switch
- skirt
- model

Why would you want it to be printed before the second color, rather than before the first?

karabas, 2016-12-06:
But cura 15.04 prints normally as expected. 
The problem is that for one mixing nozzle ( cyclop) tower must be much bigger ( currently 30mm) So skirt does not cover this need.
Correct order I believe must be:
1 skirt 
2 model 1st color 
maybe skirt
3 tower 2nd color
maybe skirt but 2nd color
4 model 2nd color
5 next layer
  
Correct?

BagelOrb, 2016-12-06:
I just checked and the order you describe is the order in which Cura 2.4 does it.
This issue is fixed in the next release.

closed 2016-12-06 by BagelOrb