Ultimaker/Cura #1218: Ultimaker 3 Extended - Profile Manger not showing any profiles Cura 2.3.1

ianlpowell, 2016-12-05:
Initially Profile Manger show no profiles. Also after creating a new one no profiles is shown in the manger.

ianlpowell, 2016-12-05:
Here is the log file
[cura.zip](https://github.com/Ultimaker/Cura/files/631753/cura.zip)

nickthetait, 2016-12-05:
Might help to add a screenshot of this problem

ianlpowell, 2016-12-05:
Sequence - Creating new profile and result. 

[image]
[image]
[image]

fieldOfView, 2016-12-05:
So you added a custom material. That is useful information to reproduce your bugreport. Care to share your material file(s)?

ianlpowell, 2016-12-06:
Do need all of them or just the added ones.
BTW I did not see any profiles just after I installed Cura but at this time I did not know what to expect.
I am running Cura in Windows 10 ver 1607.

Ghostkeeper, 2016-12-06:
It filters the known profiles based on the "Material Type" field in the material editor. If you changed this to something unknown, Cura won't find any quality profiles to display.

We changed this in 2.4 to display the default profiles (changing only the layer height) if this happens.

closed 2016-12-06 by Ghostkeeper