Ultimaker/Cura #1219: Cura 2.3.1 crashes while adding printer

Mamu1986, 2016-12-05:
I have a problem with starting the cura 2.3.1. software.

After choosing the printer cura crashes and win7 tells me that cura does not work an ends...

This problem exists when I choose the Ultimaker 2 or Ultimaker 2+.

When I choose Ultimaker Original or a so called custom printer, cura didn't crash.

I Use Win 7 x64 German Cura does not work in English or German language.

I have also tried out to Instal and run Cura as admin.

How can I run Cura with mit UM2+???

nallath, 2016-12-06:
Could you post your logs?

Ghostkeeper, 2016-12-06:
We might want to consider catching a few permission errors here and there if this turns out to be an admin issue again.

Mamu1986, 2016-12-06:
Thank you for your answers.

You find the logs at the end of that comment.

anti Virus and Firewall is switched off.

I tried to run cura Again. crash... When I click on the Details of the error, it's called: BEX64

So I think that I may have a 64 bit compatibility problem?!?

For the BEX64 Error ther're some fix clips on Youtube. (https://www.youtube.com/watch?v=HxfmgjCC-_c) -> I tried it like this:

rigth click on Cura.exe -> Propertys ->Compatibility -> Click on compatibility Mode and choose Vista Servicepack 2 -> Close

system properties -> advanced -> performance / settings -> data execution prevention -> turn on DEP for all... -> add -> click on Cura.exe -> ERROR Atributes can not be settet on 64 bit executables.

.... fix does not work =(

After that I tried the 32 bit Cura Software, -> crash while choosing UM2+ printer...

I have testet the 32 bit Cura Software too. crash...

I wonder that I don't have a crash while starting cura... It just crashes when I Add the UM2 Printers... I havn't found something like this in several communities... =(

Do you have any more ideas?

The Logs of stderr from cura are:

2016-12-06 20:48:31,080 - INFO - D:\2.3\build\inst\lib\python3.4\site-packages\UM\PluginRegistry.py (loadPlugin [63]): Loaded plugin ConsoleLogger
2016-12-06 20:48:31,142 - INFO - C:\Program Files\Cura 2.3.1\plugins\CuraEngineBackend\CuraEngineBackend.py (__init__ [40]): Found CuraEngine at: C:\Program Files\Cura 2.3.1\CuraEngine.exe
2016-12-06 20:48:31,142 - INFO - D:\2.3\build\inst\lib\python3.4\site-packages\UM\PluginRegistry.py (loadPlugin [63]): Loaded plugin CuraEngineBackend
2016-12-06 20:48:31,158 - INFO - D:\2.3\build\inst\lib\python3.4\site-packages\UM\PluginRegistry.py (loadPlugin [63]): Loaded plugin CuraProfileReader
2016-12-06 20:48:31,189 - INFO - D:\2.3\build\inst\lib\python3.4\site-packages\UM\PluginRegistry.py (loadPlugin [63]): Loaded plugin CuraProfileWriter
2016-12-06 20:48:31,283 - INFO - D:\2.3\build\inst\lib\python3.4\site-packages\UM\PluginRegistry.py (loadPlugin [63]): Plugin Doodle3D uses an incompatible API version, ignoring
2016-12-06 20:48:31,314 - INFO - D:\2.3\build\inst\lib\python3.4\site-packages\UM\PluginRegistry.py (loadPlugin [63]): Loaded plugin 3MFWriter
2016-12-06 20:48:31,345 - INFO - D:\2.3\build\inst\lib\python3.4\site-packages\UM\PluginRegistry.py (loadPlugin [63]): Loaded plugin OBJReader
2016-12-06 20:48:31,376 - INFO - D:\2.3\build\inst\lib\python3.4\site-packages\UM\PluginRegistry.py (loadPlugin [63]): Loaded plugin OBJWriter
2016-12-06 20:48:31,407 - INFO - D:\2.3\build\inst\lib\python3.4\site-packages\UM\PluginRegistry.py (loadPlugin [63]): Loaded plugin STLReader
2016-12-06 20:48:31,439 - INFO - D:\2.3\build\inst\lib\python3.4\site-packages\UM\PluginRegistry.py (loadPlugin [63]): Loaded plugin STLWriter
2016-12-06 20:48:31,470 - INFO - D:\2.3\build\inst\lib\python3.4\site-packages\UM\PluginRegistry.py (loadPlugin [63]): Loaded plugin FileLogger
2016-12-06 20:48:31,501 - INFO - D:\2.3\build\inst\lib\python3.4\site-packages\UM\PluginRegistry.py (loadPlugin [63]): Loaded plugin GCodeProfileReader
2016-12-06 20:48:31,532 - INFO - D:\2.3\build\inst\lib\python3.4\site-packages\UM\PluginRegistry.py (loadPlugin [63]): Loaded plugin GCodeWriter
2016-12-06 20:48:31,579 - INFO - D:\2.3\build\inst\lib\python3.4\site-packages\UM\PluginRegistry.py (loadPlugin [63]): Loaded plugin ImageReader
2016-12-06 20:48:31,626 - INFO - D:\2.3\build\inst\lib\python3.4\site-packages\UM\PluginRegistry.py (loadPlugin [63]): Loaded plugin LayerView
2016-12-06 20:48:31,657 - INFO - D:\2.3\build\inst\lib\python3.4\site-packages\UM\PluginRegistry.py (loadPlugin [63]): Loaded plugin LegacyProfileReader
2016-12-06 20:48:31,690 - INFO - D:\2.3\build\inst\lib\python3.4\site-packages\UM\PluginRegistry.py (loadPlugin [63]): Loaded plugin LocalFileOutputDevice
2016-12-06 20:48:31,737 - INFO - D:\2.3\build\inst\lib\python3.4\site-packages\UM\PluginRegistry.py (loadPlugin [63]): Loaded plugin MachineSettingsAction
2016-12-06 20:48:31,768 - WARNING - D:\2.3\build\inst\lib\python3.4\site-packages\UM\Preferences.py (getValue [70]): Tried to get the value of non-existing setting cura/active_mode.
2016-12-06 20:48:31,768 - INFO - D:\2.3\build\inst\lib\python3.4\site-packages\UM\PluginRegistry.py (loadPlugin [63]): Loaded plugin PerObjectSettingsTool
2016-12-06 20:48:31,815 - INFO - D:\2.3\build\inst\lib\python3.4\site-packages\UM\PluginRegistry.py (loadPlugin [63]): Loaded plugin PostProcessingPlugin
2016-12-06 20:48:31,877 - INFO - D:\2.3\build\inst\lib\python3.4\site-packages\UM\PluginRegistry.py (loadPlugin [63]): Loaded plugin RemovableDriveOutputDevice
2016-12-06 20:48:31,924 - INFO - D:\2.3\build\inst\lib\python3.4\site-packages\UM\PluginRegistry.py (loadPlugin [63]): Loaded plugin SliceInfoPlugin
2016-12-06 20:48:31,955 - INFO - D:\2.3\build\inst\lib\python3.4\site-packages\UM\PluginRegistry.py (loadPlugin [63]): Loaded plugin SolidView
2016-12-06 20:48:32,002 - INFO - D:\2.3\build\inst\lib\python3.4\site-packages\UM\PluginRegistry.py (loadPlugin [63]): Loaded plugin CameraTool
2016-12-06 20:48:32,096 - INFO - D:\2.3\build\inst\lib\python3.4\site-packages\UM\PluginRegistry.py (loadPlugin [63]): Loaded plugin MirrorTool
2016-12-06 20:48:32,159 - INFO - D:\2.3\build\inst\lib\python3.4\site-packages\UM\PluginRegistry.py (loadPlugin [63]): Loaded plugin RotateTool
[... cut ...]
2016-12-06 20:49:57,962 - DEBUG - C:\Program Files\Cura 2.3.1\plugins\CuraEngineBackend\CuraEngineBackend.py (_terminate [186]): Killing engine process
2016-12-06 20:49:57,967 - DEBUG - C:\Program Files\Cura 2.3.1\plugins\CuraEngineBackend\CuraEngineBackend.py (_terminate [186]): Engine process is killed. Received return code 1
2016-12-06 20:49:58,076 - DEBUG - D:\2.3\build\inst\lib\python3.4\site-packages\UM\Mesh\MeshData.py (approximateConvexHull [305]): approximateConvexHull(target_count=1024) Calculating 3D convex hull took 0.020000934600830078 seconds. 265 input vertices. 265 output vertices.

ianlpowell, 2017-01-10:
I have a similar problem (Windows 10 ver 1607 Ultimaker 3 Extended)

If I clear the platform after successful print and try an load a new model Cura crashes.

DanielJoyce, 2017-01-19:
I have this exact same problem on Kubuntu. Cura crashes upon trying to add a prusa i3 printer.

**uname -a**

Linux crusoe 4.8.0-34-generic #36-Ubuntu SMP Wed Dec 21 17:24:18 UTC 2016 x86_64 x86_64 x86_64 GNU/Linux

**/etc/lsb-release**

DISTRIB_ID=Ubuntu
DISTRIB_RELEASE=16.10
DISTRIB_CODENAME=yakkety
DISTRIB_DESCRIPTION="Ubuntu 16.10"

Ghostkeeper, 2017-01-22:
Mamu's log says nothing about a crash :(

Daniel's problem seems different, since it happens when adding a printer rather than when clearing the build plate.

ianlpowell, 2017-01-24:
Hi, this problem seems to have gone away. I have noticed that with Windows 10 when a update is waiting to be executed strange thing happen to some apps particularly ones involving peripherals. So this may have been the cause., although I don't know if this was the case.

Ghostkeeper, 2017-01-24:
Adding a printer involves very few peripherals though.

Ghostkeeper, 2017-01-31:
Our change control board just closed this for being unable to reproduce it.

mariolpantunes, 2018-03-01:
I have the same error with version 3.2.1 and running on Linux.
The final output of the program is the following:
Fatal Python error: Segmentation fault

Thread 0x00007f7dbbfff700 (most recent call first):
  File "/usr/lib64/python3.6/site-packages/UM/Backend/Backend.py", line 155 in _storeStderrToLogThread
  File "/usr/lib64/python3.6/threading.py", line 864 in run
  File "/usr/lib64/python3.6/threading.py", line 916 in _bootstrap_inner
  File "/usr/lib64/python3.6/threading.py", line 884 in _bootstrap

Thread 0x00007f7ddcff9700 (most recent call first):
  File "/usr/lib64/python3.6/site-packages/UM/Backend/Backend.py", line 147 in _storeOutputToLogThread
  File "/usr/lib64/python3.6/threading.py", line 864 in run
  File "/usr/lib64/python3.6/threading.py", line 916 in _bootstrap_inner
  File "/usr/lib64/python3.6/threading.py", line 884 in _bootstrap

Thread 0x00007f7dddffb700 (most recent call first):
  File "/usr/lib/cura/plugins/USBPrinting/USBPrinterOutputDeviceManager.py", line 86 in _updateThread
  File "/usr/lib64/python3.6/threading.py", line 864 in run
  File "/usr/lib64/python3.6/threading.py", line 916 in _bootstrap_inner
  File "/usr/lib64/python3.6/threading.py", line 884 in _bootstrap

Thread 0x00007f7dde7fc700 (most recent call first):
  File "/usr/lib64/python3.6/threading.py", line 299 in wait
  File "/usr/lib64/python3.6/site-packages/zeroconf.py", line 1716 in wait
  File "/usr/lib64/python3.6/site-packages/zeroconf.py", line 1317 in run
  File "/usr/lib64/python3.6/threading.py", line 916 in _bootstrap_inner
  File "/usr/lib64/python3.6/threading.py", line 884 in _bootstrap

Thread 0x00007f7ddeffd700 (most recent call first):
  File "/usr/lib64/python3.6/threading.py", line 299 in wait
  File "/usr/lib64/python3.6/site-packages/zeroconf.py", line 1716 in wait
  File "/usr/lib64/python3.6/site-packages/zeroconf.py", line 1181 in run
  File "/usr/lib64/python3.6/threading.py", line 916 in _bootstrap_inner
  File "/usr/lib64/python3.6/threading.py", line 884 in _bootstrap

Thread 0x00007f7ddf7fe700 (most recent call first):
  File "/usr/lib64/python3.6/site-packages/zeroconf.py", line 1102 in run
  File "/usr/lib64/python3.6/threading.py", line 916 in _bootstrap_inner
  File "/usr/lib64/python3.6/threading.py", line 884 in _bootstrap

Thread 0x00007f7ddffff700 (most recent call first):
  File "/usr/lib64/python3.6/threading.py", line 299 in wait
  File "/usr/lib64/python3.6/threading.py", line 551 in wait
  File "/usr/lib/cura/plugins/UM3NetworkPrinting/NetworkPrinterOutputDevicePlugin.py", line 330 in _handleOnServiceChangedRequests
  File "/usr/lib64/python3.6/threading.py", line 864 in run
  File "/usr/lib64/python3.6/threading.py", line 916 in _bootstrap_inner
  File "/usr/lib64/python3.6/threading.py", line 884 in _bootstrap

Thread 0x00007f7deddab700 (most recent call first):
  File "/usr/lib/cura/plugins/RemovableDriveOutputDevice/RemovableDrivePlugin.py", line 61 in _updateThread
  File "/usr/lib64/python3.6/threading.py", line 864 in run
  File "/usr/lib64/python3.6/threading.py", line 916 in _bootstrap_inner
  File "/usr/lib64/python3.6/threading.py", line 884 in _bootstrap

Thread 0x00007f7def7fe700 (most recent call first):
  File "/usr/lib64/python3.6/threading.py", line 295 in wait
  File "/usr/lib64/python3.6/threading.py", line 426 in acquire
  File "/usr/lib64/python3.6/site-packages/UM/JobQueue.py", line 86 in _nextJob
  File "/usr/lib64/python3.6/site-packages/UM/JobQueue.py", line 116 in run
  File "/usr/lib64/python3.6/threading.py", line 916 in _bootstrap_inner
  File "/usr/lib64/python3.6/threading.py", line 884 in _bootstrap

Thread 0x00007f7deffff700 (most recent call first):
  File "/usr/lib64/python3.6/threading.py", line 295 in wait
  File "/usr/lib64/python3.6/threading.py", line 426 in acquire
  File "/usr/lib64/python3.6/site-packages/UM/JobQueue.py", line 86 in _nextJob
  File "/usr/lib64/python3.6/site-packages/UM/JobQueue.py", line 116 in run
  File "/usr/lib64/python3.6/threading.py", line 916 in _bootstrap_inner
  File "/usr/lib64/python3.6/threading.py", line 884 in _bootstrap

Thread 0x00007f7dfc868700 (most recent call first):
  File "/usr/lib64/python3.6/threading.py", line 295 in wait
  File "/usr/lib64/python3.6/threading.py", line 426 in acquire
  File "/usr/lib64/python3.6/site-packages/UM/JobQueue.py", line 86 in _nextJob
  File "/usr/lib64/python3.6/site-packages/UM/JobQueue.py", line 116 in run
  File "/usr/lib64/python3.6/threading.py", line 916 in _bootstrap_inner
  File "/usr/lib64/python3.6/threading.py", line 884 in _bootstrap

Thread 0x00007f7e16840700 (most recent call first):
  File "/usr/lib64/python3.6/threading.py", line 295 in wait
  File "/usr/lib64/python3.6/threading.py", line 426 in acquire
  File "/usr/lib64/python3.6/site-packages/UM/JobQueue.py", line 86 in _nextJob
  File "/usr/lib64/python3.6/site-packages/UM/JobQueue.py", line 116 in run
  File "/usr/lib64/python3.6/threading.py", line 916 in _bootstrap_inner
  File "/usr/lib64/python3.6/threading.py", line 884 in _bootstrap

Current thread 0x00007f7e2993e700 (most recent call first):
  File "/usr/lib64/python3.6/site-packages/UM/Qt/ListModel.py", line 62 in setItems
  File "/usr/lib64/python3.6/site-packages/cura/Settings/ExtrudersModel.py", line 212 in __updateExtruders
  File "/usr/lib64/python3.6/site-packages/UM/Qt/QtApplication.py", line 241 in exec_
  File "/usr/lib64/python3.6/site-packages/cura/CuraApplication.py", line 743 in run
  File "/usr/bin/cura", line 141 in <module>
Segmentation fault

Ghostkeeper, 2018-03-23:
The function that the traceback refers to:

    def setItems(self, items):
        self.beginResetModel()
        self._items = items
        self.endResetModel()
        self.itemsChanged.emit() #<--- This is line 62.

So it appears that something in Qt was listening to this itemsChanged signal and crashed when processing it.

And you also got the same reproduce steps? Happened upon adding a printer?

mariolpantunes, 2018-03-26:
It happens when I open the application, it starts searching for a printer in the network and it breaks.

Ghostkeeper, 2018-04-19:
So something like this would be your reproduce steps:

* Add an Ultimaker 3 printer.
* Connect to your printer.
* Restart Cura. --> Crash

Right?
I can't reproduce the problem in this way though.

mariolpantunes, 2018-04-21:
I do not restart. During the network search cura just crashes. Is there a log that i can sendo?

alekseisasin, 2018-05-08:
yes, you can find a log in your user folder:

    %APPDATA%\cura\<Cura version>\cura.log (Windows), or usually C:\Users\\<your username>\AppData\Roaming\cura\<Cura version>\cura.log
    $USER/Library/Application Support/cura/<Cura version>/cura.log (OSX)
    $USER/.local/share/cura/<Cura version>/cura.log (Ubuntu/Linux)

ianpaschal, 2018-05-28:
Not reproducible.

closed 2018-05-28 by ianpaschal