Ultimaker/Cura #1220: Shape in shape support issue

peteruithoven, 2016-12-06:
When moving (stacking) shapes on top of each other and they contain a shape inside them there is unnecessary support generated. I've created a small example, including a shape with no shape inside where this problem doesn't occur. 
[image]
[image]
[image]
[Support issue example.zip](https://github.com/Ultimaker/Cura/files/634116/Support.issue.example.zip)

I'm using Cura 2.3.1 from the ppa. 

Let me know if I can provide more information.

nickthetait, 2016-12-06:
Thanks for writing such a good good test case! You deserve some :cake:

Is this reproducible using a stock profile, only adjusting the support overhang angle right?

peteruithoven, 2016-12-06:
I rechecked this with stock profiles. You get a similar result when selecting the Ultimaker 2 or Ultimaker 3 and the Normal Quality profile and enabling support.  
[support issue.curaprofile.zip](https://github.com/Ultimaker/Cura/files/635298/support.issue.curaprofile.zip)

peteruithoven, 2016-12-07:
If you guys know of some kind of workaround I'd love to know.

Ghostkeeper, 2016-12-08:
This reproduces for me too with `support test.stl`. I sliced on Ultimaker 2 with default Normal Quality settings except that support is enabled.

It seems to be caused by Support Join Distance. Setting it to 0 removes the effect (so there's your workaround). Setting it to 1.5 removes the effect too. Around 1.8 it seems to join the pieces. Support Join Distance is 2mm by default.

If you set the brim line width to 1mm, you can see in layer view that it fits 4 brim lines in between, so the distance between the inner part and the outer part seems to be about 4mm. But due to the resolution of the circle, the distance is smaller almost everywhere.

The "Support Join Distance" setting is implemented by expanding the support polygons by the value of the setting, then insetting them with the same value. Perhaps the "fix" would be to double the setting's value towards the user and then halve the effect, so that it is clear from the value of the setting that it can bridge a distance of 4mm by default (instead of 2mm).

BagelOrb, 2016-12-12:
The problem is, however, that there is a small gap in between the two pieces which requires support, but the gap is too small, so no support is generated between the models, while the ungenerated support does have influence via the support join distance.

I'm not sure whether all support areas are removed in the gap - there might be unprintably small areas designated to be filled with support, which cannot be printed because they re too small. If that's the case then the bug is a logical by-product of your settings. If it's not the case then we might need to tweak the algorithm such that support areas which are later removed never introduce new support areas.

peteruithoven, 2016-12-12:
@BagelOrb But that gap is filled with a support interface? Or are you referring to another gap? And then again this support shouldn't be generated above the empty areas (which triggers more support in this case)?

BagelOrb, 2016-12-12:
Oh of course; you are printing support interface. In that case this issue is a logical by-product of your settings.

Try turning the Support Join Distance down if you don't want the current behaviour.

However, PVA works best if it has a non-interupted line to the build plate. Without the support in the middle there is a high chance that the interface won't start printing on top of the model.

closed 2016-12-12 by BagelOrb

peteruithoven, 2016-12-12:
Please explain why it's logical it prints supports in areas it clearly doesn't need support? (The empty areas in between these shapes).

> However, PVA works best if it has a non-interupted line to the build plate. Without the support in the middle there is a high chance that the interface won't start printing on top of the model.

True, found that out already: https://ultimaker.com/en/community/34824-pva-on-pla-adhesion-stacking-prints But using the `Support Horizontal Expansion` setting for this is a valid workaround.

BagelOrb, 2016-12-12:
I'm not saying it's logical that it prints support in places where it doesn't need to; I'm saying that where support is generated is a logical consequence of the settings and this case doesn't disprove that. We can't optimize the settings to be perfect for any print.

Support Join Distance is used to merge separate support pillars into one big pillar, which is sturdier and sometimes even requires less material. Your situation is a logical consequence of having a support join distance of 2mm. Just tweak that setting down to get your print the way you want.

peteruithoven, 2016-12-12:
Ah I finally get your point, when I scale up the example model this doesn't happen.

peteruithoven, 2016-12-12:
At 200%:
[image]
Or probably when the distance between the support interfaces is more than 2mm (the default Support Join Distance).

BagelOrb, 2016-12-12:
It doesn't happen when the support interfaces are farther apart than 4mm.
It should be 2mm, that is indeed a little bug; it does twice.

<https://cloud.githubusercontent.com/assets/523210/21100601/49934f06-c075-11e6-8e97-2e5068c659be.png>

<https://github.com/Ultimaker/Cura/issues/1220#issuecomment-266428508>, or mute
the thread