Ultimaker/Cura #1223: multiple models are sliced wrong (same models)

jeroensomers, 2016-12-07:
When duplicating a model multiple times (or just opening the same model multiple times), some models aren't sliced correctly.
In this particularly case its the hook at the side of the model.
[image]
When deleting the model that is good it turn's the other models in 1 good and 1 wrongly sliced.
[image]

I attached the .stl file and my profile settings so you can try it yourself.

[stl and profile.zip](https://github.com/Ultimaker/Cura/files/636687/stl.and.profile.zip)
I'm using Cura 2.3.0

I tried reloading the model through different ways (dragging and dropping it in and through the menu).
I tried turning the model and mirroring the model, nothing seems to help.

peteruithoven, 2016-12-07:
This also happens with the default Normal Quality profile (using Ultimaker 2). 
It's probably causes by overlapping shapes:
[image]
For some reason it doesn't happen when I set the `Print sequence` setting to `One at a time`. 
For some reason it seems to be influenced by the `Print sequence` setting, changing that sometimes "fixes it", even in `All at Once` mode, but this is very unpredictable.

fieldOfView, 2016-12-07:
Switching to One at a Time and back probably pushed the objects further from eachother.

Ghostkeeper, 2016-12-08:
Also, One At A Time mode basically runs most of the process in the engine separately for every object and then appends the g-code in sequence.

Ghostkeeper, 2016-12-08:
I can't reproduce this :( Not with the provided models and settings. It prints the little hook properly on all three copies.

I had scaled the mesh to 50%.

peteruithoven, 2016-12-08:
So unfair ;) 
We scaled them to 30% but that can't hardly matter. 
You tried Normal Quality / Jeroen's profile? Print sequence All at once? 

When I separate them in 3 stl's (ear, outer ring and inner gears) and merge those in Cura it seems to work without issues.

Ghostkeeper, 2016-12-08:
Yup, using the profile in the .zip.

peteruithoven, 2016-12-12:
Ultimaker 2, Normal Quality, model 30% scaled, then duplicated 3 times. Moving the third shape 2mm to the right:
[image]
[image]

Looks like something is wrong with the `Union Overlapping Volumes` feature when models are positioned to close to each other? 

File: [Gear BearingV1 (7).stl.zip](https://github.com/Ultimaker/Cura/files/646159/Gear.BearingV1.7.stl.zip)

Ghostkeeper, 2016-12-12:
I did get this reproduced at home on 2.3.0 (I tried it at home previously as well). Though it didn't position the models in the exact same places for me, it still occurred.
Perhaps I should try a bit harder when I'm at work with 2.4.

This is the build plate I used using Ultimaker 2, Normal Quality defaults:
[UM2_Gear BearingV1 (7).zip](https://github.com/Ultimaker/Cura/files/647343/UM2_Gear.BearingV1.7.zip)
I would've exported a workspace, but of course that feature is not available yet in this ancient version ;)

Ghostkeeper, 2016-12-14:
I still can't get this to reproduce in 2.4. We did change a few things around with how the outlines are generated though. Perhaps this fixed it.

peteruithoven, 2016-12-19:
Awesome work guys! It seems to be fixed in 2.4 (from the ppa):
[image]

(Also really digging the settings filter and multiply features)

fieldOfView, 2016-12-19:
> (Also really digging the settings filter and multiply features)

(thanks)

closed 2016-12-20 by Ghostkeeper